uzi18/sterownik #3: mapowanie czujnikow

uzi18, 2016-02-13:


uzi18, 2016-03-03:
mapowanie z ecoal zrobione ... linijka w konfiguracji:
MAPUJ_Z_ECOAL=1
powoduje przypisanie wszystkich czujnikow z ecoal/bruli do odpowiednich w pamieci sterownika
pozostalo dodac dallasy ... i kazdego z osobna

bzykbzyk, 2016-03-06:
program v19 nie działa u mnie. W ogóle nie zgłasza się sterownik. Nie odpowiada na ping. V 17 OK. Zauważyłem pewną nieścisłość: w index (v19) w sekcji czujników jest ['tCO','tCWU1','tCWU2','tSpaliny','t4D'], a piszesz powyżej TCWU i nie ma wcale t4D.

uzi18, 2016-03-06:
Poprawilem byl problemz kompilatorem, po zmianie ten sam programdziala poprawnie.
czujniki nie dzialaja jeszcze na stronie, na wykresie powinno byc ok.

bzykbzyk, 2016-03-06:
Dalej brak komunikacji. Dioda 13 świeci ciągle.

uzi18, 2016-03-07:
Dioda 13 jest od komunikacji z ecoal.

bzykbzyk, 2016-03-07:
Dioda 13 miała migać wolno albo szybko a skoro świeciła ciągle to wydedukowałem,  że brak komunikacji. v22 wrzucona do procesora. Strona się zgłasza, na wykresie widać zakres temp -1 do 1, więc nie widzę wykresu...

uzi18, 2016-03-07:
wklej co masz w pliku config.txt i co pokazuje wywolanie /dallas no i moze screenshot jak to wyglada

bzykbzyk, 2016-03-07:
Kawałek /dallas

DALLAS 0 ID:2890B65005000091 p:1 o:0 t:212
DALLAS 1 ID:28188A4B050000D3 p:1 o:0 t:213
DALLAS 2 ID:2843EE4F05000005 p:1 o:0 t:211
DALLAS 3 ID:0000000000000000 p:0 o:0 t:0
DALLAS 4 ID:0000000000000000 p:0 o:0 t:0
A poniżej cofig.txt

Adres IP:
IP1=192
IP2=168
IP3=2
IP4=50

Adres MAC:
MAC1=00
MAC2=01
MAC3=02
MAC4=03
MAC5=04
MAC6=05

Parametry dmuchawy:
DM_OFFSET = 25
DM_INWERSJA_PWM = 0
DM_MODULACJA = 1

Parametry modulacji (gdy DM_MODULACJA = 1):
DM_ROZRUCH_CZAS = 90
DM_ROZRUCH_MOC = 80
DM_PLUS_CZAS = 10
DM_PLUS_MOC = 10
DM_DOPALANIE_CZAS = 60
DM_DOPALANIE_MOC = 40
DM_TLO_MOC = 30
tPIEC = DS18[28188A4B050000D3]
tPOWROT = DS18[2890B65005000091]
tPODAJNIK = DS18[2843EE4F05000005]

Kiedy odczytywany jest przez program plik config.txt? Czy po zapisaniu przez stronę do sterownika uaktualniany jest też txt?
Czasem zdarzało mi się, że po odczytaniu konfiguracji przez stronę zawartość w okienku i w txt była niezgodna z początkową (była krótsza o kilkanaście znaków).
[image]

Stan17, 2016-03-07:
$ cat resetujpiec
#!/bin/sh
avrdude -patmega2560 -cwiring -P/dev/ttyACM0 -b115200 -V -U
lfuse:r:/dev/null:i

uzi18, 2016-03-07:
zapisz takie cos w pliku config.txt:
`Adres IP:
IP1=192
IP2=168
IP3=2
IP4=50

Adres MAC:
MAC1=00
MAC2=01
MAC3=02
MAC4=03
MAC5=04
MAC6=05

Parametry dmuchawy:
DM_OFFSET = 25
DM_INWERSJA_PWM = 0
DM_MODULACJA = 1

Parametry modulacji (gdy DM_MODULACJA = 1):
DM_ROZRUCH_CZAS = 90
DM_ROZRUCH_MOC = 80
DM_PLUS_CZAS = 10
DM_PLUS_MOC = 10
DM_DOPALANIE_CZAS = 60
DM_DOPALANIE_MOC = 40
DM_TLO_MOC = 30
TPIEC = DALLAS[28188A4B050000D3]
TPOWROT = DALLAS[2890B65005000091]
TPODAJNIK = DALLAS[2843EE4F05000005]
koniec`

closed 2016-03-09 by uzi18