uzi18/sterownik #5: lista standardowych czujnikow

uzi18, 2016-02-17:
CO
CWU
POWROT

bzykbzyk, 2016-02-18:
ZEW
KOMIN
PIEC
CWU1
PODAJNIK
PODLOG

uzi18, 2016-02-18:
CWU1 czy CWU2 ?

uzi18, 2016-02-18:
Czym sie rozni CO od PIEC?

whitemar, 2016-02-18:
Witam Was,
Proponowałbym powielić czujnik CO, jako zabezpieczenie w razie jego uszkodzenia: CO1 i CO2. Pełniłyby tę samą funkcję i były ulokowane w tym samym miejscu, a awaria jednego z nich (zbyt duża różnica wskazań CO1 i CO2) wymuszałaby alarm o konieczności wymiany. Drugi czujnik w tym czasie zapobiegłby unieruchomieniu kotła.

bzykbzyk, 2016-02-18:
CO to temperatura wody za czterodrogowym (jeśli jest) albo na rurociągu "zasilanie" a PIEC to bezpośrednio w pochwie pieca. 
Nie trzeba powielania (każdy czujnik może ulec uszkodzeniu, trzeba by wszystkie powielać). Niebezpieczny jest niekontrolowany wzrost temp pieca. Oprócz czujników konieczne zastosowanie termostatów (jak na schemacie): bezpośrednio w pochwie 90-95st odcinający zasilanie dmuchawy i podajnika - piec zgłosi awarię brak paliwa, albo dwóch nc i no - jeden rozłączy zasilanie, drugi wyśle sygnał do sterownika, albo jednego z przekaźnikiem przełączającym.

uzi18, 2016-02-19:
Podoba mi sie ta opcja ... PIEC na piecu a obwody jako CO1,CO2 itp.

bzykbzyk, 2016-02-19:
Jeśli czujnikom dodatkowym można nadać nazwy, to oczywiście PODLOG i KOMIN wyrzuć z podstawowych, no chyba że dodatkowe to tylko DS.

uzi18, 2016-02-19:
bzyk nie ma/nie bedzie ograniczen w tej kwestii po to bedzie wlasnie mapowanie :)

uzi18, 2016-02-25:
aktualnie zrobilem takie:
  t_piec,
  t_powrot,
  t_podajnik,
  t_zew,
  t_cwu,
  t_spaliny,
  t_t1,
  t_t2,
  t_t3,
  t_t4,
  t_t5,
  t_t6,
  t_t7,
  t_t8

bzykbzyk, 2016-02-25:
No bo już zaczęły pleśnieć mi płytki a synek już czaił się żeby je spożytkować....

uzi18, 2016-02-25:
Bez przesady moja 2 maluchow tez absorbuja, a najchetniej by sie tym wszystkim bawili :)

closed 2016-03-09 by uzi18