uzi18/sterownik #8: dmuchawa AC

uzi18, 2016-02-18:


bzykbzyk, 2016-02-20:
uzi18, jesteś pewny że avt-3113 nadaje się do sterowania silników asynchronicznych? Producent twierdzi, że przeznaczony jest do komutatorowych.

uzi18, 2016-02-25:
jestem pewny, w tym tygodniu przyjda płytki to zloze i sprawdze ale to jest praktycznie to samo co Stan zmontowal z 1613 i fragmentu z 3113 wiec nie ma innej mozliwosci

uzi18, 2016-09-12:
Podsumowując moje testy: 
avt3113 - działa poprawnie : sygnał 0V-0% 5V-100%, mozliwosc regulacji minimalnych i maksymalnych obrotów oraz biegu jałowego, przy 100% obroty troszke pływają wiec warto zjechac troszke z obrotów maksymalnych potencjometrem na module

avt1613-zmodyfikowany wg Stan - działa poprawnie poza tym ze wymaga odwroconego sygnału (5V-0% 0V-100%) i u mnie na 0% dmuchawa sie nie potrafi zatrzymać.

triak-optotriak+detekcja zera - w trakcie testów

bzykbzyk, 2016-09-16:
avt3113 niestety niedostępny

uzi18, 2016-09-16:
Aktualizacja:

avt1613-zmodyfikowany wg Stan - działa poprawnie, 
wg. mnie kiepskie mozliwosci regulacji wzgledem 3113, prawdopodobnie mozna to jeszcze dopracowac
wczesniej zle dzialala bo u2008b mialem w podstawce,  najlepiej przylutowac go bezposrednio do plytki

avt3113- jest niedostepna i producent nie ma zamiaru jej wznowic do sprzedazy

mark3k3, 2016-09-18:
Ogólnie, moduł AVT1613 działa dość poprawnie, jednak PWM nie może wysterować wentylatorem na MAX. MAX obroty nie są nam potrzebne, dlatego takie wykonanie modułu wykonawczego sterowania dmuchawą jest jak najbardziej akceptowalne i przydatne.

bzykbzyk, 2016-10-01:
Dla dmuchawy nie jest potrzebne 100%, ale już np dla pompy tak.
Chciałbym w przyszłości wykorzystać Lucjana również do sterowania panelami słonecznymi i obrotami pompy: w zależności od różnicy temperatur przed i za panelami. Muszę zbudować układ na wzór avt3113, skoro lepiej steruje.

uzi18, 2016-10-01:
Moze dla pompy zakres regulacji na avt1613 tez wystarcza.

bzykbzyk, 2016-10-01:
Muszę zatem zbudować jakieś stanowisko do pomiaru obrotów pompy i sprawdzić jak się zachowuje. Nawet jak by nie było max 100%, można w ostateczności zewrzeć regulator stykiem przekaźnika.

uzi18, 2016-10-01:
Mozna ustawic/wyregulowac plytke 1613 aby byl zakres od 40%-100%.

uzi18, 2016-10-06:
Moge potwierdzic ze swojej strony ze dmuchawka AC sterowana 
bezposrednio z Lucjana w  konfiguracji - detektor zera, optotriak, triak - smiga az milo ...

bzykbzyk, 2016-10-07:
Mój piec na razie w gotowości do palenia. Przestawiam trochę szpeje w kotłowni więc brak czasu. Czekam na moduł triaków.
Skoro 1679 działa zadowalająco (w całym zakresie obrotów?) to rezygnujemy z 1613?
Dlaczego sterujesz triakami z wyjść analogowych a nie cyfrowych (dmuchawa - wiadomo)?
Uaktualniłem schemat (bez przekaźników na razie, info w wątku o przekaźnikach).

uzi18, 2016-10-08:
Jezcze troche i zrobimy porzadki z wyjsciami.
Wyjscie na opto to jakiekolwiek zwykle wyjscie cyfrowe.

Czekam na wasze testy z moc3020, na razie nikt nie potwierdzil poprawnej
pracy, a 1613/3113 jest dobra alternatywa.

bzykbzyk, 2016-10-08:
Przy porządkowaniu wyjść, rozpatrz przeniesienie gdzieś sygnałów DO i CLK termopary, dla triaków "cyfrowych" można by przeznaczyć wyjścia 46 -55 - zrobić kabelek 10 żyłowy (wtyk - gniazdo) kanda -  pewnie i prosto podpiąć moduł triaków (nie przerabiany) do mega jedną taśmą. W tej chwili wykorzystujemy już 8 wyjść do sterowań, lada chwila trzeba dołożyć kolejny moduł triaków. Jakaś pompa np do solarów też ładnie by hulała ze zmienną prędkością zależną od temp.

ravikjab, 2016-10-11:
uruchomiłem dmuchawę na MOC3020 i po wstępnych testach praca dmuchawy WPA01 jest wielce obiecująca. Jest większa moc względem AV1613. Dmuchawa w trybie modulacji bez problemów zmienia predkość. 
Więcej informacji przekaże po dłuższych testach.

uzi18, 2016-10-18:
ktoś jeszcze testował?

uzi18, 2016-11-05:
v0.1.0.66
        - sterowanie triakiem dmuchawy (A13) zdublowane na pin 32

Docelowo będziemy musieli przenieść się z pinów Ax (zwłaszcza dmuchawa)z prostej przyczyny - procesor ma jakiś bug powodujący zakłócenia na wyjściach przełączanych z "wysoką" częstotliwością.

Tym samym czekam na potwierdzenie ze dmuchawa sterowana optotriak/triak działa już
poprawnie po podłączeniu pod pin 32.
A13 z czasem ... będzie to testów czujników np. KTY81-210.

mark-markowski, 2016-11-06:
Potwierdzam prawidłowe działanie dmuchawy na pinie nr. 32. Modulacja działa płynnie i precyzyjnie.

uzi18, 2016-11-27:
Rozumiem,że działa więc zamykam.

closed 2016-11-27 by uzi18