uzi18/sterownik #15: dmuchawa cyklicznie

Stan17, 2016-03-17:
Dodać w trybie letnim cykliczną pracę dmuchawy.

mark3k3, 2016-10-02:
Nie jestem pewien celowości wprowadzenia tego rozwiązania. Na okres letni, gdy kocioł stoi wygaszony u siebie dokładnie zamykam wlot powietrza przez wentylator (folia), a do kociołka wkładam żel absorbujący wilgoć i jest dość dobrze. Raz nie zamknąłem kotła na lato i w środku miałem sporo skroplonej wody i multum rdzy. Ścianki kotła są mocno zimne i szybko wykrapla się na nich woda, gdy tylko jest przepływ powietrza przez kocioł.

bzykbzyk, 2016-10-03:
Moje zdanie jest identyczne jak mark3k3. Przewietrzanie pieca to tylko chwilowe jego osuszenie, dostarcza za to świeże powietrze z nową porcją pary wodnej. Ja również proponuję nie używany szczelnie zamknąć. Ja opróżniam z węgla, czyszczę i smaruję zużytym olejem silnikowym. W dymnice upycham szmaty nasączone olejem, w palenisku i zasobniku kładę tacki z wapnem hydratyzowanym.

uzi18, 2016-10-03:
Pale caly rok wiec nie mam tego problemu, Stan myslisz ze moge temat
zamknac?

Stan17, 2016-10-03:
Ale dlaczego uważacie że skoro Wy robicie inaczej to to rozwiązanie jest nie dobre ?
Ja piec odstawiam na lato ale muszę go mieć w gotowości do odpalenia, i opróżnianie całkowite, korki w dymnicach itp. nie wchodzi w grę.

uzi18, 2016-10-03:
Ok wiec na przyszly rok przetestujemy taka funkcje, chyba ze chcesz to
robic teraz?
Tylko przypomnij sie abym mogl przygotowac program.

bzykbzyk, 2016-10-04:
Ależ Stan! Nie jest moim celem przekonywanie Ciebie na siłę do mojego zdania! Przecież i tak zrobisz jak zechcesz. Po prostu wiem z doświadczenia, że pozostawianie otwartego i nie wyczyszczonego pieca węglowego w czasie jego nieużywania (nawet kilkudniowego) niezbyt mu służy. Dostające się do niego świeże powietrze z kotłowni powoduje po kilku dniach pojawienie się kropel wody kondensującej się na zimnych blachach, sadza i popiół potęguje to zjawisko, bo dodatkowo chłonie wilgoć z powietrza. Wystarczy po kilku dniach postoju przesunąć dłonią po blachach wewnątrz, dłoń jest mocno wilgotna. Kwas ze spalin i woda - idealne do biodegradacji tak pożądanej w czasach dbania o środowisko. Co prawda w instrukcjach pieców węglowych, producenci zalecają pozostawianie otwartych drzwiczek podczas dłuższych postojów sprzętu, ale czy naprawdę są to gadki w interesie użytkownika?  Nie sądzę, coby nasze (i producenta) interesy były zbieżne... Moim zdaniem, po każdym wygaszeniu pieca (gdy planujemy kilka dni nie palić) powinno się go wyczyścić z sadzy i popiołu, przesmarować olejem i szczelnie zamknąć, niekoniecznie uszczelniać dymnice, choć u mnie to tylko 5 otworów fi 40. Producent zaleca opróżnienie zasobnika, ale komu by się chciało.
Ja wlewam niewielką (do 0,5l) porcję oleju po każdym załadowaniu zasobnika na czubek węgla. Przenika do ślimaka i smaruje go, zmniejszona podatność na korozję, opory, zużycie energii i hałas. Można też użyć karbosmaru za jakieś 30zł za litr, ale nie wiem czy zawiera coś więcej niż olej i pył węglowy.
Ależ mi się zrobił elaborat...

uzi
Żeby nie było niedomówień, nie mam nic przeciw wprowadzeniu timera do przedmuchwania pieca podczas postoju, oczywiście dodatkowo wszystkie pompy, np raz na dobę po kilkanaście sek - albo najlepiej jako programator, tak jak planujesz dla pompy CWU  PROG_CYRK. Przecież kto chce niech używa.

Stan17, 2016-10-05:
No widzisz a u mnie to jednak się sprawdza, ale nigdy nie twierdziłem że sprawdzi się każdemu i w każdych warunkach.
Trick z olejem również stosuję :)