uzi18/sterownik #20: zabezpieczenie www

bzykbzyk, 2016-03-20:
Dodać zabezpieczenie strony

bzykbzyk, 2016-03-20:
Zabezpieczenie strony

uzi18, 2017-09-07:
Jestem w trakcie dopisywania tej funkcji. 
@Stan17 @mipurel @mark3k3 @ravikjab dodam na poczatek podstawowe hasla a pozniej mozliwosc zdefiniowania swojego loginu i hasła.
Macie jakies pomysly w tym temacie?
Domyslnie wlaczyc to zabezpieczenie? 
API www tez zablokowac hasłem?

mipurel, 2017-09-08:
W API na pewno należy zabezpieczyć funkcje nieinformacyjne czyli dać na hasło np. reset, set,  modulacje i configreload. Większość użytkowników nie ma potrzeby zabezpieczania dlatego sądzę, że domyślnie można zostawić wyłączone. W moim przypadku chciałbym, żeby wszyscy mieli podgląd do www (bez config.txt) a podgląd i zmiana była możliwa tylko przez osobę uprawnioną.

uzi18, 2017-09-24:
dodano w 0.1.0.129.4

closed 2017-09-24 by uzi18