uzi18/sterownik #21: Reakcja pieca na alarm (brak paliwa,  pożar zasobnika, awaryjna temp w kotłowni, przegrzanie silnika podajnika)

bzykbzyk, 2016-03-20:
Reakcja pieca na alarm (brak paliwa,  pożar zasobnika, awaryjna temp w kotłowni, przegrzanie silnika podajnika)

uzi18, 2016-03-20:
jak powinien zareagowac na te alarmy?
to juz mamy: pożar zasobnika, awaryjna temp w kotłowni

bzykbzyk, 2016-03-20:
Brak paliwa:
przez dłuższy, zadany czas (30 min?) brak przyrostów temp pieca, tylko spadek - wyłączenie pieca, alarm na stronie i jednym z wyjść. Po każdym wzroście temp - reset odliczania czasu. Gdy w podtrzymaniu - czas braku nie jest liczony.

Pożar zasobnika: w tej chwili jest tylko awaryjny wyrzut, przyda się jeszcze alarm na stronie i na wyjściu. Jeśli po wyrzucie paliwa temp podajnika nie wzrasta to piec pracuje dalej.

Jaka jest reakcja na alarm w kotłowni? Powinien być stop pieca i alarm na stronie i wyjściu.

Przegrzanie silnika podajnika: stop pieca do czasu spadku temp silnika do zadanej wartości, komunikat na stronie "sprawdź silnik, opory toczenia, łożyska, kondensator", alarm na wyjściu. Jeśli w tym czasie pożar zasobnika - praca cykliczna podajnika, jeśli czas stygnięcia dłuższy od podtrzymania - chwilowe uruchomienie aby zapobiec wygaszeniu.
Wszystkie te alarmy wymagają ręcznego skasowania.

Stan17, 2016-03-21:
Alarmów nie mamy na razie wcale, sprawa do dorobienia ale czy jako pilna ?
Mamy tylko reakcję na zdarzenia.
Brak paliwa - przydatne, ale wykorzystałbym do tego czujnik temp. spalin bo reakcja byłaby znacznie szybsza. Jeżeli w grzaniu podczas nadmuchu temperatura spalin nie rośnie znaczy że piec wygasł.

Pożar zasobnika - wiadomo.

A co to jest alarm w kotłowni ? Jakiś alarm z czujnika zewnętrznego ?

Przegrzanie silnika - ciekawy pomysł, jak do tej pory wymieniłem chyba trzy kondensatory i faktycznie taki alarm uchroniłby mnie przed skutkami niekontrolowanego postoju silnika.

bzykbzyk, 2016-03-21:
Skoro Lucjan (vel UziStan - to tylko z przekory...) "sam pali" to są różne priorytety poszczególnych nowych czy modyfikacji starych opcji, to moim zdaniem decyzja głównego programisty. 
Nie mam czujnika spalin, dlatego kombinuję na okrętkę, docelowo na pewno wyposażę - to najlepsza opcja.

Alarm w kotłowni to temperatura z czujnika umieszczonego na suficie nad piecem, nie za blisko czopucha.

W moim silniku podajnika (kupiłem używany dwa sezony piec) po kilkudziesięciu minutach pracy na biegu luzem włączało się zabezpieczenie termiczne, co nie powinno się wcale zdarzać. Rozebrałem całkowicie silnik, wymieniłem łożyska i dokładnie go wyczyściłem ale to nie przyniosło spodziewanego efektu. Kondensator był 8mikroF, nie miałem takiego, dałem 2x3,75. Efekt ten sam. Stwierdziłem, że silnik ma jakieś uszkodzenia wew i dlatego się grzeje, ale ponieważ pracuje cyklicznie to jakiś czas pożyje. Jakiś czas temu Uzi18 też zwrócił mi uwagę na kondensator, zacząłem kombinować jeszcze raz i ku mojemu zdziwieniu idealnie pracuje na połowie fabrycznej pojemności, czyli 3,75. Kondensator antystroboskopowy z oprawy świetlówkowej. Gdyby był za mały prąd w uzwojeniu z kond, silnik nie osiągał by pełnej mocy i się przegrzewał. A mój praktycznie wcale się nie grzeje, po godzinie jest jedynie lekko ciepły, dużo ciszej pracuje.

uzi18, 2016-03-21:
Nie ma potrzeby chowac czujnika w silniku, bo obudowa doskonale przewodzi ciepło

bzykbzyk, 2016-03-21:
Ale wentylator skutecznie chłodzi przekłamując, wewnątrz puszki przyłączeniowej jest ciepło i przytulnie

uzi18, 2016-03-21:
mysle ze o to nie ma co sie martwic, jesli wzrosnie temperatura to czujnik to wychwyci.

uzi18, 2016-09-19:
co do kondensatora - im wiekszy tym prad przy rozruchu bardziej rosnie i dlatego wywala termik :)