uzi18/sterownik #22: detekcja przeciążenia silnika podajnika

bzykbzyk, 2016-03-21:
Pomiar prądu silnika podajnika - przeciążenie - włączenie pracy nawrotnej

uzi18, 2016-03-21:
zrobimy testy na ACS712 w wersji 5A, moze uda sie wylapac tez zerwanie zawleczki

Stan17, 2016-04-15:
Testy zrobione, ACS712 i AC-MAX-30A, zerwanie zawleczki jest raczej nie do wykrycia ale przeciążenie na pewno tak.

bzykbzyk, 2016-04-17:
AC-MAX-30A ma jakąś elektronikę? Czy to tylko transformator i na wyjściu prąd zmienny do wykorzystania? Jakieś schematy do wrzucenia?

uzi18, 2016-04-17:
Ma rezystor i diody zabezpieczajace na wyjsciu trafa, wykonania sa rozne.

bzykbzyk, 2016-04-17:
uzi18
Ha, czujny jesteś. Napisałeś odpowiedź na moje pytanie o konstrukcji AC-MAX-30A, które po kilku minutach usunąłem bo znalazłem ją w sieci. Skoro twierdzisz że są różne wykonania, to nie jest dobrze. To utrudnia unifikację. Może sam coś zbuduję. Te 30A wydaje mi się za dużo do  naszego zastosowania, skoro mamy mierzyć prądy w okolicach 1A. 30 amperowe mają przekładnię 1800:1 muszę zrobić przekładnik o mniejszym przełożeniu. Widzę też w sieci 5, 10, 15A 
Stan17,
dlaczego twierdzisz, że zerwanie zawleczki nie do wykrycia? Przecież to musi być większy prąd niż przeciążenie.
W jakim układzie to testujesz? Daj go, zbuduję to samo i sprawdzę

uzi18, 2016-04-17:
Efekt na kazdym czujniku jest ten sam, one sie glownie rezystorem roznia, w zaleznosci od pradu, kalibracje najpewniej i tak kazdy bedzie musial zrobic

Stan17, 2016-04-18:
Dlaczego tak myślę ? Ponieważ kiedyś to zmierzyłem i różnica między prądem normalnej pracy a prądem pracy z odłączonym ślimakiem wynosiła około 0,1 A.

Tu masz przykład którym się posłużyłem.
http://akademia.nettigo.pl/power_meter/

micro-jr, 2016-04-18:
nie prościej w okolicach zawleczki zamontować jakiś czujnik obrotu ślimaka?
silnik chodzi, czujnik kręci - OK
silnik chodzi - czujnik nie kręci - zerwana zawleczka
silnik nie chodzi, czujnik nie kręci - OK
silnik nie chodzi, czujnik kręci - ??

Stan17, 2016-04-18:
Ale głównym celem nie jest wykrycie zerwania zawleczki, to tylko dodatek.
Podstawową sprawą tutaj jest wykrycie przeciążenia zanim zostanie zerwana zawleczka, wykonanie ćwierć/pół obrotu ślimaka wstecz i powrót do normalnych obrotów.

micro-jr, 2016-04-18:
no to pytanie jaki jest pobór podczas przeciążenia, a nie pracy bez zawleczki.
bo jeśli do ścięcia zawleczki potrzeba 0,1A więcej, to słabo to widzę... chociaż z drugiej strony, pewnie, przed zerwaniem zawleczki, obroty silnika mocno spadają - a zatem po ilości obrotów (silnika, a nie ślimaka) można by reagować  - jeśli podajemy prąd, a ilość obrotów gwałtownie spadła to mamy awarię.

uzi18, 2016-04-18:
Nie o to chodzi, ten czujnik daje nam wiecej informacji, ktore odpowiednio
analizujac mozemy dobrze wykorzystac

Stan17, 2016-04-18:
Ale ja nie to napisałem.
To oczywiste że do ścięcia potrzeba o wiele większego prądu.
Chcemy wykryć przeciążenie i zareagować zanim nastąpi ścięcie zawleczki.

Chodzi mi o to że tak mała różnica między prądami pracy z zawleczką i bez, nie pozwoli nam w sposób absolutnie pewny stwierdzić czy po wzroście prądu w przeciążeniu i jego nagłym spadku podajnik pracuje z zawleczką czy bez.

bzykbzyk, 2016-04-20:
Mierzenie obrotów silnika (w celu detekcji ścięcia zawleczki) jest moim zdaniem bezcelowe, bo z uwagi na duże przełożenie przekładni spadek obrotów będzie (raczej) minimalny. Wszystkie występujące na rynku podobne układy bazują na pomiarze prądu silnika. Widziałem układ z pomiarem różnicy obrotów pomiędzy ślimakiem a reduktorem (czujnik Halla plus magnesy). Jeśli różna od 1 = ścięta zawleczka.  Był to układ tylko do detekcji ścięcia. Tylko po co taka wiedza (za parę stów - bo tyle tamten kosztował)? Skoro sterownik podaje sterowanie na silnik a temperatura nie rośnie to znaczy, że brak paliwa lub zerwana zawleczka. Jedno i drugie wiąże się z wizytą w kotłowni.
Cała zabawa w tym wątku ma zakończyć się budową układu detekcji przeciążenia (zablokowania) podajnika grożącego uszkodzeniem szpejów wartości co najmniej tysiaka a nie zawleczki wartej ćwierć złocisza.
Jeśli uda się zbudować skuteczny układ, to z czystym sumieniem wymienimy zawleczki na utwardzone śruby ponieważ rola zawleczki zostanie przejęta przez ten układ. Takie rozwiązanie jest stosowanie przez kilku producentów "elektronicznych zawleczek".

**uzi18,**
Dopiszże w wolnej chwili kawałek kodu do pomiaru prądu układem ACS712 5A z wizualizacją na wykresie (łatwiej będzie analizować pracę podajnika).

uzi18, 2016-04-20:
Stan tez o to prosil, aby dopisac jakis kod do tego czujnika, zobacze co da sie zrobic.

micro-jr, 2016-04-21:
Jeżeli silnik jest w trybie pracuj, ale się nie obraca - znaczy ze doszło do zacięcia.
czy silnik ma pracować wiemy - bo to My decydujemy.
pozostaje wykryć, czy silnik/Przekładania się obraca. 

zakładając, ze silnik obraca się szybciej niż podajnijk i jesteśmy w stanie określić z jakimiś interwałem o ile się obraca, jeśli w zadanym czasie obrót nie nastąpił, a silnik pracował - mamy zacięcie silnika.

nie wiem jaka jest czułość czujników obrotu "za 30 pln" i czy da się taki założyć przed przekładnią, bo tam są większe obroty - szybciej powinna nastąpić zmiana - mamy mniejsze opóźnienie.

Stan17, 2016-04-21:
Na połaczeniu  silnika z motoreduktorem nie ma gdzie go zamontować, jedyne miejsce to wykrycie obrotu wentylatora silnika.
Jeśli chcesz to próbuj takiego rozwiązania, ja jednak pozostanę przy wykryciu przeciążenia :-)

uzi18, 2016-04-21:
Panowie przetestujmy wpierw jedno rozwiazanie z czujnikiem pradu, pozniej sie zastanowimy czy jest wystarczajace.

Moim zdaniem czujnik śruby na ślimaku daje za rzadko informacje o obrocie, zakładając ze kilka sekund na cykl podawania to tez mało na 1/2 lub cały obrót ślimaka.

My chcięlibyśmy mieć informacje z wyprzedzeniem zanim zetniemy zawleczkę, a pozniej zareagować kilkusekundowa zmiana obrotów silnika.

bzykbzyk, 2016-04-28:
Mój ślimak wykonuje 10 obrotów w czasie 8min. Silnik (wg tabliczki zn) - 1430 obr/min. Daje to przełożenie ok 1150:1. 
Moim zdaniem nawet po zablokowaniu ślimaka spadek obrotów silnika będzie minimalny, raczej nie mierzalny domowymi sposobami. 

Ostatnio chciałem zmierzyć przyrost prądu silnika po obciążeniu (zablokowaniu) ślimaka.
Mam sprawny układ testowy: silnik z przekładnią zdemontowany z pieca przeznaczonego na złom. Ślimak ucięty szlifierką, ponieważ została ścięta zawleczka i przekładnia "zblokowała się" ze ślimakiem. Przygotowałem szlifierką "kikut" ślimaka, by pasował do klucza płaskiego 22. Założyłem klucz tak, aby po kilku sekundach obrotu zablokował się opierając o korpus, podłączyłem arduino z ACS712 5A. Z uwagi na brak zawleczki sądziłem, że klucz na kikucie ślimaka będzie wystarczający do przezwyciężenia blokady z przekładnią.
Wyniki:
Prąd max > ok 10% od znamionowego.
Na szczęście stałem poza trajektorią przelotu części klucza.
Szkoda tylko zniszczonego klucza vanad (miałem jeszcze zwykły, stalowy, mogłem użyć jego...).

Wnioski:
Układ przeciążenia trzeba testować podczas normalnej pracy napędu, z zawleczką fabryczną.
Rejestrować przeciążenia, czekać na zerwanie zawleczki. Odczytać wartości zarejestrowane w czasie zerwania, ustawić kilka % niższą, jako wyzwalającą układ do pracy nawrotnej.

ACS712 5A nie jest zbyt precyzyjny. Z uwagi na charakter pomiaru (cz Halla) nie jest odporny na zakłócające pole magnetyczne.
Program w arduino z którego korzystam podczas obciążenia wskazuje podobny prąd do zmierzonego przyrządem, a bez obciążenia to 0,25-0,3A. Bez sensu.
Proszę o kod z którego Wy korzystacie, może tu jest przyczyna?

uzi18, 2016-09-08:
Jestem na etapie składania modułu pomiarowego, niebawem dam znac jakie efekty :)

uzi18, 2016-09-18:
Mysle ze bedzie wybor miedzy termometrem silnika a modulem pomiaru obciazenia.
bzyk: jak mozesz to wrysuj detektor zera sieci za bezpiecznikiem od peryferiow ... 
bedzie dzieki temu zarazem detektorem przepalenia bezpiecznika :)

bzykbzyk, 2016-09-19:
Za bezpiecznikiem AVT1679? Mogę wrysować, ale czy da się to fizycznie wykonać?

uzi18, 2016-09-19:
U siebie przylutowalem przewod za bepiecznikiem

bzykbzyk, 2016-09-19:
Ok, wrysuję mostki zamiast bezpieczników i jeden wspólny do wszystkiego prócz zasilacza.