uzi18/sterownik #23: Możliwość pracy z kilkoma plikami konfiguracyjnymi

bzykbzyk, 2016-03-21:
Można by zdefiniować kilka plików np z parametrami dla różnych typów paliwa i czytać w razie potrzeby.

Stan17, 2016-03-21:
Tak, o taką możliwość wielokrotnie proszono ES dla ich sterowników, sam również o to prosiłem.

uzi18, 2016-03-21:
mhm, czyli raczej opcje spalania w oddzielnym pliku ..., a jak sobie wyobrazacie wybor tych plikow?

bzykbzyk, 2016-03-22:
Moim zdaniem wszystko powinno być w jednym pliku. Program przy uruchamianiu czyta parametry z pliku config.txt. Na stronie www mamy możliwość wyboru pliku przez najzwyklejsze otwórz plik lub zapis bieżącej konfiguracji z pamięci pod dowolną nazwą w celu łatwego orientowania się co tam mamy. Podczas zapisu program powinien zapisać ją pod nową nazwą i jednocześnie zastąpić config.txt . Gwarantowało by to po ponownym uruchomieniu odczyt ostatniej konfig bez kombinacji z pamiętaniem przez program, co było ostatnio. Jak coś pochrzanimy w konf, wystarczy na karcie zapisać poprawny plik config.txt i wszystko ok.

uzi18, 2016-03-22:
zobacze co da sie zrobic, plikami bedziemy zarzadzac poprzez www, to pewne.
moze czesc parametrow zachowam w eeprom - te kluczowe np. ip,mac,maska sieci i hasla do strony.
wtedy w konfigu beda one (te dane) potrzebne tylko w wypadku pierwszego uruchomienia

bzykbzyk, 2016-03-22:
No właśnie, jak działa ten tajny program uzi18'ego. Kiedy czyta config np.

uzi18, 2016-03-22:
konfig jest czytany zaraz po resecie

bzykbzyk, 2016-03-23:
Czyli również po uruchomieniu?

uzi18, 2016-03-23:
zgadza się

Stan17, 2016-11-21:
Podbijam temat, przydała by się taka możliwość do testów steownika.

uzi18, 2016-11-21:
Mozesz miec kilka konfiguracji na komputerze i wysylac raz ta raz tamta.
Dla ulatwienia skopiowac sobie i zmodyfikowac skrypt do wysylania configu
aby konkretny plik wysylal.