uzi18/sterownik #24: Wyjścia sterujące przekażnikami

bzykbzyk, 2016-03-24:
Przydała by się wersja programu sterująca modułem przekaźników zamiast triaków ew 
mogą to być dodatkowe wyjścia z odwróconą polaryzacją bo popularne moduły wymagają sterowania wysokim poziomem.
Albo zmienna w programie zmieniająca poziom na poszczególnych wyjściach.

uzi18, 2016-03-24:
pokaz co to za moduly - link?

bzykbzyk, 2016-03-25:
Etam będę Ci linkował, obrazki pokażę. Występują pod nazwą "moduł przekaźnikowy arduino" od jednego do ośmiu, z optoizolacją.
[image]
[image]

Stan17, 2016-03-25:
Nie wiem czy przekaźniki to dobry pomysł, to jest dość zawodny element.

uzi18, 2016-03-25:
Ale te przekazniki maja styk NO i NC wystarczy ze obciazenie podlaczysz do drugiego i bedziesz mial odwrocony sygnal na wyjsciu

micro-jr, 2016-03-25:
Nie odwrócony. Tylko normalnie zwarty lub normalnie otwarty. Jeśli potrzebujesz żeby coś chodziło przez 90% czasu to lepiej sterować wtedy czasem wyłączenia. W piaszczysta wtedy t a ka rzecz w normalnie zwarty. A w sterowniku ustawiasz jako aktywny port 0 wówczas jest to zwarte kiedy przekaźnik nie pracuje.podanie jedynki na sterowniku zmieni stan przekaźnika na prace i rozewrze obwód. Oszczędza to przekaźnik.  

Pokazane na zdjęciu układy to opto przekaźniki.  Ich zaletą jest izolacja galwaniczna obwodów. Jeśli nie ma błędu ma płytce zwiększa odporność systemu na przecięcia.

bzykbzyk, 2016-03-25:
Przekaźniki może i zawodne, ale kilka razy zdarzyło mi się, że po zwarciu nawet styki się nie zgrzały a triak to kosz.
Nie można korzystać z NC styku, bo po wyłączeniu sterownika z jakiegokolwiek powodu (choćby reset) wszystkie obwody zostaną zasilone, co np dla zaworu 4d może być zabójcze.

uzi18, 2016-03-25:
Dlatego przed triakami daje sie bezpiecznik :)

bzykbzyk, 2016-03-27:
Bezpiecznik nawet szybki nie zabezpiecza w 100% triaka. Nieraz zadziałał bezp i triak kosz.

uzi18, 2016-03-27:
Przed wszystkim sie nie zabezpieczysz niestety.

Stan17, 2016-03-27:
Nie wiem jak wy, ale ja wolę w razie awarii wymienić triaka niż przekaźnik wlutowany.
Ale co kto woli :-)

uzi18, 2016-03-27:
Bzyk pozwol ze na razie zajme sie rozwinieciem tego co juz mamy, temat dopisze do ulepszen, na ten czas mozna zastosowac sprzetowa inwersje, moze znajdziesz gotowy modul?

Dodanie inwersji w tym momencie komplikuje troszke sprawe, bo mamy kilka innych problemow do rozwiazania.

Dodaje temat do ulepszen, zgloszenie zostawiam otwarte

uzi18, 2016-03-27:
Mozna tez sprobowac w prosty sposob przerobic modul przekaznikow aby sygnal nie musial byc odwracany ...

uzi18, 2016-03-27:
#232d8548631ae3c2d7efac8581f78b6af6d77a82

uzi18, 2016-03-27:
dodane w 0.1.0.39 - prosze o testy

bzykbzyk, 2016-03-29:
Dziękuję za wersję dla przekaźników. Pewnie, że można by te moduły przerobić, ale kłóciło by się to z zasadą, aby korzystać jak najbardziej z gotowych rozwiązań. 
 Z uwagi na połączenia, dobrzy by było zaprogramować parzyste wyjścia sterujące 24, 26, itd.

uzi18, 2016-03-29:
parzyste naprawde cos dadza, bo nie wiem jak to rozumiec?

Mozesz dorysowac na schemacie modul rtc "ds1307+24c32", podlaczone vcc,gnd,scl,sda

bzykbzyk, 2016-03-29:
Chodzi o podłączenie do pinów arduino elementów wykonawczych i czujników. Można użyć "Złącze ARK do płytki stykowej", ma wyprowadzenia gold pinów w jednym rzędzie. A to właśnie parzyste piny 22 - do DS oraz 24, 26 itd do przekaźników. Docelowo (jeśli to nie byłby problem) lepiej by było wykorzystać jednak już te od triaków, (coby nie dublować wyjść) zmieniając im tylko polaryzację (może jakieś zmienne do ustawienia w config).
Jeśli chodzi o moduły przekaźnikowe, to gapa ze mnie, występują też sterowane wysokim potencjałem.
Schemat uaktualnię.

uzi18, 2016-03-29:
To fajny pomysl z tymi zlaczami jednak jest to dosc ograniczone jesli chodzi o mozliwosci podlaczenia, za to wydaje mi sie ze widzialem gdzies gotowce na cala plytke ...

bzykbzyk, 2016-03-29:
Są na całą Mega, ale nie miałem tego w łapkach, więc trudno określić czy się nada.
Niestety Lucjan w tej chwili na biurku, jestem na etapie szycia mu nowego garnituru, czyli obudowy. 
Czekam na gniazda C13, moduł pomiaru prądu nie może dotrzeć (przesilenie na poczcie), regulator ac dmuchawy czeka na złożenie, i ... trochę mało czasu, więc wszystko opornie idzie. Każdy nowy wsad testuję na biurku, symulacja pieca to lampa biurkowa i zapalniczka. Od kilkunastu dni w ruchu, pracuje stabilnie. Nie grzeję cwu węglem (na razie - w planie zasobnik samoróbka, szukam zbiornika hydroforowego z demobilu) więc z uwagi na temperatury zewnętrzne piec wygaszony.

uzi18, 2016-03-29:
dopisalem troche rzeczy do wiki ...

bzykbzyk, 2016-03-30:
Ta płytka to do testów jedynie. Niestety nie ma wyprowadzonych zacisków śrubowych na D22 do D53

uzi18, 2016-03-30:
Ma za to pola gdzie mozna takie lub inne zlacze przylutowac

bzykbzyk, 2016-03-30:
No niestety, nie jest to takie proste, bo złącze śrubowe typu ARK ma raster 5mm a w Arduino jest 2,5mm, stad moja propozycja  "Złącze ARK do płytki stykowej"

uzi18, 2016-03-31:
Ok, uwolnie teraz zlacza analogowych wejsc i zrobimy na parzystych od 24 stan niski aktywny a na nieparzystych wysoki aktywny, chociaz to marnotrastwo ale na razie musi tak byc.
Miałem plan aby 24-55 uzyc do innych celów.

bzykbzyk, 2016-04-01:
Nie można by nie dublować, ale z użyciem parametrów w config załatwić sprawę? Wiadomo, że kawałek kodu musiałbyś skrobnąć i nie musi być to od razu.
Na schemacie jeszcze nie wszystko ok, wejścia sterujące przekaźników trzeba spolaryzować do plusa. 
Jakiś duży mi ten plik teraz wyszedł...
Znalazłem schield do Mega z rozszyciem D24-50 na śrubowe,
[https://www.tindie.com/products/duanebenson/universal-screw-block-proto-shield-system-for-arduino/](url)
lub
[image]

ale cena zabójcza. Muszę coś innego wymyślić np w oparciu o płytkę uniwersalną.

ravikjab, 2016-09-22:
Zauważyłem chyba błąd w sterowaniu gdyż przy ustawieniu TPIECA na 50 st i histereza na 2 st. to przy spadaniu temp podajnik właczył się przy 47,9 st (czyli prawidłowo), natomiast gdy temp rosła to już po przekroczeniu 48 st podajnik przestał się właczać a powinien pracować do uzyskania nastawionej temperatury 50 st.
W tym momencie korzystam z przekaźników.

uzi18, 2016-09-22:
Dodalem to jako błąd, do weryfikacji.

uzi18, 2016-09-25:
Dodalem nastepne wyjscia dla przekaznikow w wersji v.0.1.0.47

bzykbzyk, 2016-10-07:
Uwolnij wyjścia cyfrowe 24 - 30. Nie ma sensu sterować przekaźnikami stanem niskim, skoro są dostępne moduły sterowane wysokim a nawet mające możliwość konfiguracji stanu przełączania (załączenie przekaźnika wysokim lub niskim).
Jak ktoś będzie chciał jednak przekaźniki, może je włączyć zamiast triaków.
[image].

uzi18, 2016-10-08:
Ok przemyslimy sprawe, kto uzywa jeszcze zwyklego modulu przekaznikow?