uzi18/sterownik #28: Uwagi do www

bzykbzyk, 2016-04-03:
Błędnie opisany poziom zasobnika. Jest "Poziom podajnika".
Po dodaniu paliwa, ilość w "Poziom podajnika" się nie zwiększa, tzn wygląda jakbyśmy zasypywali pusty zasobnik. 
Ile to jest "Do pełna"?. Są różne zasobniki.

Można ustawić "Aktualizuj co:" jako liczbę ujemną.

Stan17, 2016-04-03:
U mnie poziom się zwiększa lub zmniejsza zależy ile "wsypię", zasyp działa inaczej niż w sterach ES, tzn gdy wpisujesz ile kilogramów to nie oznacza ile wsypałeś tylko ile masz w zasobniku czyli łączna ilość to co wsypałeś plus to co tam ewentualnie zostało.

Do pełna określasz w configu
> #czas w sekundach potrzebny do oproznienia pełnego zasobnika
> PODAJNIK_CZAS_DO_OPROZNIENIA = 69000

Ale w konfigu jest niekonsekwencja, w opisie mamy to usunąć 
> #aktualny czas pozostaly do oproznienia zasobnika w sekundach
> #po uruchomieniu zliczamy od tej wartosci (domyslnie 50% zasobnika)
> PODAJNIK_CZAS_PODAJNIKA = 34500
A w aktualnym konfigu ciągle to jest.

uzi18, 2016-04-03:
Podajnik_czas_podajnika sluzy teraz tylko do ustawienia wartosci poczatkowej, generalnie nie ma potrzeby tego uzywac i niebawem chyba to wyrzuce.
Pozostawienie tej linijki spowoduje ze po resecie czas znow bedzie wpisany taki jak w parametrze a raczej tego noe chcemy ;)
Dosypywanie zrobie, moze nawet dzis.

bzykbzyk, 2016-04-03:
A o co tu chodzi:
> Jeśli ustawię czas podawania na 10, to po "Zasypano" 100kg 
mam "Poziom podajnika" 869,6%, 
[image]
a po ustawieniu czasu na 200 - to samo 100kg daje "Poziom podajnika" 54,3%
(teraz 43%!)
[image]

>Można ustawić "Aktualizuj co:" jako liczbę ujemną.[image]

uzi18, 2016-04-03:
Pokaz zawartosc pliku config.txt

bzykbzyk, 2016-04-03:
#czas w sekundach potrzebny do oproznienia pe�nego zasobnika
PODAJNIK_CZAS_DO_OPROZNIENIA = 69000
#aktualny czas pozostaly do oproznienia zasobnika w sekundach
#sta�a zalezna od opa�u w g./min.
PODAJNIK_PREDKOSC_PODAWANIA =200
zabezpieczenie przed zbyt wysoka temperatura pieca
KOCIOL_T_MAX = 90

temperatura zalaczenia pomp
PIEC_T_MIN = 0

tryb pracy pieca (palnika) - zgodny z ECOAL,RR,RRM,...
PIEC_ALGORYTM = RR
PIEC_ZADANA = 70
PIEC_HISTEREZA = 2

parametry algorytmu RR, czas podtrzymania w minutach, reszta w sekundach:
RR_PODTRZYMANIE_CZAS = 10
RR_PODTRZYMANIE_PODANIE = 10
RR_PODTRZYMANIE_POSTOJ = 30
RR_PODTRZYMANIE_MOC = 40

RR_PRACA_PODANIE = 2
RR_PRACA_POSTOJ = 5
RR_PRACA_MOC = 23

tryb pracy pompy co (ECOAL,BRULI,ZIMA,LATO)
CO_TRYB = ZIMA
algorytm pracy pomy CO (CIAGLY,CYKLICZNY)
CO_ALGORYTM = CIAGLY

tryb pracy pompy cwu (ECOAL,BRULI,WLACZ,WYLACZ)
CWU_TRYB = WYLACZ
algorytm pracy pomy cwu (CIAGLY,CYKLICZNY)
CWU_ALGORYTM = CIAGLY
CWU_ZADANA = 40
CWU_HISTEREZA = 0

bzykbzyk, 2016-04-04:
Już zaskoczyłem, gdzie popełniam błąd.
W obydwu przypadkach zmiany parametru PODAJNIK_PREDKOSC_PODAWANIA miałem tą samą wartość PODAJNIK_CZAS_DO_OPROZNIENIA. Nie zrozumiałem co to za parametr PODAJNIK_CZAS_DO_OPROZNIENIA, a to przecież pojemność zasobnika przeliczona na sekundy pracy podajnika.
Moim zdaniem, ten parametr powinien być wyliczany przez program a nie wprowadzany ręcznie. 
Powinno się podać pojemność zasobnika w kg z instrukcji pieca (albo z doświadczenia - wsypując węgiel do zasobnika) i zmierzoną wartość parametru PODAJNIK_PREDKOSC_PODAWANIA.
Chociaż może niepotrzebnie kombinuję, niech Programista zdecyduje.

Brakuje mi za to na stronie czasu pracy do końca paliwa w zasobniku.

Stan17, 2016-04-04:
Ja podorabiamy różne statystyki to pewnie tak, na razie nie mamy praktycznie żadnych.
Wczoraj właśnie rozmawialiśmy na temat zrobienia pierwszej prostej statystyki zużycia dobowego węgla w okresie 7 dni.
Z tą sprawą nie ma to żadnego związku ale stanowiłoby pierwszy krok do utworzenia statystyk.

uzi18, 2016-04-06:
[image]

bzykbzyk, 2016-04-07:
Tak, teraz to już ładnie i czytelnie. Jakoś nie mogę uwierzyć w te 8,7kg na dobę Stan'a....

Stan17, 2016-04-07:
To jest wartość chwilowa, zaraz po uruchomieniu,  najpierw jest bardzo wysoka a po chwili spada mocno i dopiero po kilku godzinach pracy się stabilizuje, teraz mam około 11-12 kg

ravikjab, 2016-09-21:
Do interface www:
- przydałoby się żeby wyświetlane były ustawione wartości parametrów np. TPIEC, TCWU, TPODAJNIKDo 
- możliwość określenia na wykresie czasu prezentacji wyników (ostatnie 10 min daje słaby obraz tego co działo się np z CWU a dzięki temu można poszukać oszczędności np przez określenie czasu pracy pompy cyrkulacyjnej włączając ją wtedy gdy faktycznie jest potrzebna)

uzi18, 2016-09-21:
Probowales zmienic czestotliwosc odswiezania "aktualizuj co" na 60?

ravikjab, 2016-09-21:
Ok jest lepiej ale i tak fajnie było by gdyby można określić zakres który nas interesuje bez utraty dokładności

Stan17, 2018-02-08:
Myślę że na www kontrolki wentylator i wentylator_mod bardziej pasują do bloku status niż pomiary ?

uzi18, 2018-02-08:
racja, mozna je przeniesc :)