uzi18/sterownik #33: mozliwosc ustawenia zadanej dla CWU

uzi18, 2016-09-22:
Brak też możliwości definiowania oddzielnego progu włączania pompy CWU ( teraz chyba CO i CWU startują od jedenej wartości TPieca)

uzi18, 2016-09-23:
@ravikjab: parametr np.:` CWU_ZADANA=40`

ravikjab, 2016-09-23:
Chodzi o to żeby można było oddzielnie zdefiniować próg TPIEC od której pompa CWU będzie się włączać.
Czy pompa CWU2 to pompa cyrkulacji ? czy wówczas może pracować w określonych przedziałach czasowych np. pompa cyrkulacji będzie pracować od 6 do 8 rano na 2 min z przerwami 20 min.
I żeby można było ustawić kilka (do 5) takich przedziałów. Wszystkie parametry oczywiście do ustawienia w config.txt

uzi18, 2016-09-23:
chodzi Ci o temperature zalaczania pomp, tamto przenioslem do #36