uzi18/sterownik #36: programator pompy cyrkulacyjnej

uzi18, 2016-09-23:
@ravikjab: Czy pompa CWU2 to pompa cyrkulacji ? czy wówczas może pracować w określonych przedziałach czasowych np. pompa cyrkulacji będzie pracować od 6 do 8 rano na 2 min z przerwami 20 min.
I żeby można było ustawić kilka (do 5) takich przedziałów. Wszystkie parametry oczywiście do ustawienia w config.txt

uzi18, 2016-09-23:
CWU2 dziala u mnie jak nazwa wskazuje - do drugiego boilera

uzi18, 2016-09-23:
Oczywiscie

uzi18, 2016-09-25:
PROG_CYRK_PON=6:0-8:0,10:00-12:00

bzykbzyk, 2016-09-27:
Pozostałe dni tyg podobnie? trzy litery? Zakresów dziennie dowolna ilość?

uzi18, 2016-09-27:
to tylko propoozycja ale mysle ze tak, chyba ze macie lepsze pomysly?

uzi18, 2016-10-21:
przyklady zaprogramowanych harmonogramu:
```
poniedzialek w 2ch porach:
PROG_CYRK_PO=6:0-8:0
PROG_CYRK_PO=10:00-12:00

caly tydzien do 6:00 i od 18
PROG_CYRK_*=-6:00
PROG_CYRK_*=18:00-

PROG_CYRK_WT=1:0-2:0
PROG_CYRK_SR=2:0-3:0
PROG_CYRK_CZ=4:0-5:0
PROG_CYRK_PT=6:0-7:0
PROG_CYRK_SO=8:0-9:0
PROG_CYRK_ND=10:0-11:0

```

ravikjab, 2016-10-21:
Wygląda to bardzo czytelnie.

uzi18, 2016-11-01:
zrobione w wersji  v0.1.0.64
### Podoba Ci się projekt, kliknij i wesprzyj Lucjana: [image]

closed 2016-11-01 by uzi18