uzi18/sterownik #45: detekcja otwartego zasobnika

bzykbzyk, 2016-10-21:
Dodać detekcję otwartego zasobnika. Wymaga dołożenia krańcówki pokrywy zasobnika.
Otwarcie pokrywy - zatrzymanie wentylatora. Jeśli pokrywa zbyt długo otwarta zatrzymanie kotła i alarm.

uzi18, 2016-10-21:
naprawde mozna zapomniec zamkac klape?

bzykbzyk, 2016-10-21:
Nawet nie chodzi o pozostawienie klapy otwartej na oścież, ale nie domknięcie i nieszczelność z tym związaną. Na forum o ogrzewaniu pisano o zapaleniu się zasobnika którego przyczyną było nie domknięcie klapy i cofnięcie żaru do zasobnika. W moim fabrycznym sterowniku, zbyt wysoka temp podajnika powoduje wyłączenie dmuchawy i wyrzut paliwa do palnika (kilka prób) po których następuje zatrzymanie pieca. Ale co w przypadku gdy podczas wyrzutu zostanie zerwana zawleczka? O tym już "specjaliści" od kotła za kilkanaście tysiaków chyba zapomnieli. Celowo piszę w cudzysłowiu, zrobię Ci kiedyś kilka fotek ich rozwiązań, można się popłakać ze śmiechu.
Wydaje mi się, że krańcówka powinna mieć dwa styki: jeden przerywa zasilanie dmuchawy, drugi daje info do sterownika.
W moim kotle, gdy dorzucam paliwa (zwłaszcza gdy mieszam trociny z węglem) czasami mocno się dymi z zasobnika więc zawsze wyłączam dmuchawę (najszybciej wyjąć wtyczkę z dmuchawy)ale skoro automat...

uzi18, 2017-11-16:
Aby miec pewnosc ze klapa jest zamknieta to musialby byc uzyty czujnik cisnienia.
W moim zasobniku sa 2 zapinki, jesli jednej nie zamkniesz to moze byc ona nieszczelna.

Poza tym czy czujnik dymu nie rozwiazalby 2 rzeczy na raz?

bzykbzyk, 2017-11-16:
Czujnik dymu może rozwiązałby sprawę, ale zapylenie w zasobniku bardzo szybko skutecznie go zanieczyści. A czyszczenie czujki dymu - takiej z systemów pożarowych, chyba jedyna, która może dać jakieś miarodajne wyniki (system samokalibracji itp) jest skomplikowane i robienie tego co kilka dni...
U mnie w zasobniku jest nadciśnienie (mam układ dosuszania). Moim zdaniem jedynie krańcówki załatwią sprawę. Dwie zapinki - dwie krańcówki.

uzi18, 2017-11-16:
Myslalem o czujniku dymu w okolicy zbiornika nie wewnatrz, wskazywalby czy nie jest niedomkniety zbiornik.

Gdzie Cie wcielo na tyle czasu? :D

Stan17, 2018-02-08:
Do detekcji otwarcia można wykorzystać też czujnik poziomu.

uzi18, 2018-02-08:
Faktycznie mozna sprobowac

uzi18, 2018-02-08:
Chociaż wydaje mi sie ze @bzykbzyk mowil o kontroli domknięcia klapy.

Stan17, 2018-02-08:
Myślałem tutaj raczej o wykorzystaniu tego żeby po otwarciu klapy automatycznie przejść na sterowniku w tryb ręczny i po zamknięciu z powrotem w tryb auto.
Taki automat na czas zasypu i czyszczenia pieca :)

uzi18, 2018-02-08:
To jest ciekawy pomysl.
Lecz wpierw musi sie potwierdzic ze czujnik dziala poprawnie i nie
glupieje, bo co jesli nagle wykryje ze otwarta i wylaczy piec?

Stan17, 2018-02-09:
No tak, ale można zrobić warunki i wyniki zadziałania wrzucać w jakiś log, wtedy będzie wiadomo jak działa.
Albo chociaż jakiś alarm otwarcia klapy , to zaczniemy zbierać jakieś informacje.

WojtekGD, 2019-02-13:
Bardzo waźna i przydatna funkcja szczególnie dla biomasy a zapomnieć się zdarza zwykła krańcòwka i aktywacja w konfiguracji otwarcie przejście w tryb ręczny i alarm zamknięcie automat

uzi18, 2020-02-26:
@WojtekGD jedyne co moge dodac to ze klapa ma zwykle kilka zapinek i jesli nawet jest zamknieta to moze byc nie zapieta - niekoniecznie szczelna

WojtekGD, 2020-03-16:
Rozwiązanie nawet dla kilku zapinek jest bardzo proste - jedno wejście krańcówki NC połączone szeregowo otwarcie jeden powoduje zadziałanie alarmu i wyłączenie dmuchawy lub zatrzymanie pieca

uzi18, 2020-03-28:
ok wiec dodajmy ten blad w koncu :)

bzykbzyk, 2020-03-29:
Nic się nie dzieje. Palę Lucjanem w "uproszczonej" przez siebie "wersji", czytam na bieżąco co tam wyprawiacie na gitterze, a że nie mam nic do powiedzenia to się nie odzywam. "Uproszczenie" polega na tym, że steruję dmuchawą AC przekaźnikiem, obroty wyregulowane są  avt-1613. Zaczęło się od tego, że w czasie pierwszych prób z Twoim sterownikiem spalił mi się pewnej nocy silnik dmuchawy AC (zalałem wodą przy przeróbce instalacji i za kilka dni padł), na szybko zamontowałem dmuchawę z malucha i tak zostało. Po ponad roku i ta poszła z dymem, zamontowałem znowu AC, a że regulacja nie była odpowiednia w najniższym zakresie obrotów a było to znów nocą i nie miałem czasu dochodzić dlaczego, to zrobiłem po "swojemu" i tak zostało. Nie dymi, brak sadzy, pyłek bez spieków, to czego jeszcze chcieć.?. Palę tylko do ogrzewania, CWU gazem. Nie udało mi się dokończyć mojego bufora zamiast gazu to i nie mam dużych wymagań do sterownika. Może na dniach znów powróci sprawa z buforem, bo kolega covit-bezrobotny zaoferował się dokończyć temat.

WojtekGD, 2020-03-30:
W sumie to jak w konfgu będzie możliwość konfiguracji NC/NO to wszystko jedno ile i jakie krańcówki się podłączy czyli typ wejścia niski wysoki (NC/NO)  / tryb ręczny / opóźnienie/ alarm i teoretycznie gotowe

sargus123, 2020-09-02:
wystarczy czujnik poziomu węgla z warunkiem przy jakim ma odczytać że pokrywa jest otwarta. i przejść na tryb ręczny i było by to dobre jak ktoś wcześniej pisał  wtedy czyścić piec można.  a co do żaru i cofania po to jest czujnik podajnika co ma za zadanie przerzucić żar jeśli idzie cofka.

WojtekGD, 2020-09-02:
Tak nie do końca bo otwarty ale pełny zasobnik + zabezpieczenie przed cofką spowoduje, że po kilku próbach cały piec będzie zapchany opałem i dodatkowo przeciąg z podajnika - zatem zasobnik się nie zapali ale w piecu masz czarnobyl: paliwo + tlen i coraz wyższą TCO - nie do opanowania bez gaśnicy.