uzi18/sterownik #53: Monit o nowej wersji

uzi18, 2016-11-28:
Kto chce informacje o aktualizacji prosze dodawac sie to tego issue
@Stan17 @mark3k3 @bzykbzyk @ravikjab

uzi18, 2016-11-28:
v0.1.0.80     
        - LCD: zmiana opisu strony PODAJNIK->ZASOBNIK
        - LCD: wsparcie dla innego typu konwertera I2C->LCD

uzi18, 2016-11-29:
Test

wwpiotr, 2016-11-29:
@wwpiotr

Stan17, 2016-11-29:
dopisuję się.

wwpiotr, 2016-11-29:
dopisane ?

uzi18, 2016-11-29:
Test

uzi18, 2016-11-29:
Test 3

uzi18, 2016-11-29:
Test 123

uzi18, 2016-11-29:
Test 5

uzi18, 2016-11-29:
v0.1.0.81
    - LCD: menu alarmow - klikniecie kasuje alarm
    - dodanie do zapytania /dallas informacji czy zmapowany (m:1 - TAK)
    - zmiana sposobu obslugi alarmow

uzi18, 2016-11-29:
v0.1.0.82
    - dodanie komendy ! w terminalu - resetowanie alarmow
    - przeniesienie sterowania pompa CWU2 w jedno miejsce
    - przeniesienie sterowania pompa CWU w jedno miejsce
    - dodanie komendy a w terminalu - drukuje liste alarmow
    - dodanie komendy d do terminala - drukuje liste dallasow

uzi18, 2016-11-29:
v0.1.0.83
    - poprawki dla buzzera

uzi18, 2016-11-30:
v0.1.0.84
        - dodanie trybu recznego
        - LCD: dodanie obslugi trybu recznego
        - podajnik przeniesienie sterowania w jedno miejsce

    v0.1.0.85
        - ENKODER: dlugie wcisniecie powoduje przejscie do ekranu trybu recznego
        - LCD: zmiana opisu akcji trybu recznego
        - LCD: dodanie informacji o trybie pracy A(uto)/R(eczny) na glownym ekranie
        - poprawki dla trybu recznego podajnika

    v0.1.0.86
        - przed wyjsciem do trybu auto wylaczamy pompy poza CO
        - dodanie mozliwosci ustawiania parametrow poprzez serial: sDM_ROZRUCH_CZAS=15

uzi18, 2016-12-01:
v0.1.0.87
        - dodanie mozliwosci symulacji czujnikow zmapowanych jako SOFT[]
        - poprawienie informacji o statusie sterowania triakiem
        - dodanie informacji o braku alarmu dla komendy a

uzi18, 2016-12-13:
v0.1.0.88
        - DALLAS: dodanie informacji o ilosci bledow komunikacji
        - dodanie typu czujnika BRAK[]
        - poprawki przy monitorowaniu detektora - 4 w konsoli
        - dodanie zapytania t.json
        - Z4D: poprawka pracy przy braku czujnika t_zew dla trybu zadanej
        - pin 35 zwarty do masy zmienia rodzaj encoder-a

uzi18, 2017-01-16:
v0.1.0.89
        - RS232: wysylamy komunikat START po przygotowaniu systemu do pracy
        - RTC: zmiana sposobu wydruku informacji o czasie
        - TRIAK: zmiana sposobu pomiaru ilosci polokresow
        - dodanie wstepnej analizy pradu podajnika
        - dodanie odczytu wejsc analogowych
        - DMUCHAWA: ustawienia poczatkowe w jednym miejscu
        - DALLAS: zwiekszenie ilosci czujnikow do 16
        - DALLAS: dodano komunikat o nieudanym dodaniu czujnika

uzi18, 2017-01-20:
v0.1.0.90
      przeniesienie termostatu z pinu A6->36

    v0.1.0.91:
        - dodanie mozliwosci mapowania MAX6675K przyklad:
        TSPALIN = MAX6675[0]
        - dodanie obslugi MAX6675K pin CS - 49
        - zmiana podlaczenia pin CS z MAX31855 A7->48
        - uzywamy wlasnej biblioteki do termopar oparych na kostkach MAX

    - aktualizacja pliku polaczen

uzi18, 2017-01-21:
v0.1.0.92
    - LCD: poprawki w wyswietlaniu wartosci ujemnych dla TZEW oraz pelniejsze nazwy czujnikow
    - LCD: dodanie strony z temperatura podajnika
    - komenda t w terminalu drukuje aktualne temperatury czujnikow

uzi18, 2017-01-24:
v0.1.0.94
    - LCD: poprawka dla strony z TZEW - nie wyczyszczony znak
    - dodanie filtrowania pomiaru dla MAX6675 - mapowac z opcja [1] tj. TSPALIN=MAX6675[1]    v0.1.0.95
        - LCD: poprawne usuniecie problemow z TZEWN na wyswietlaczu
        - dodanie opcji DETEKTOR_TESTUJE_BEZPIECZNIK=1
          dla osob ktorym nie dziala detektor lub nie testuja nim bezpiecznika
          0 - wylacza bledy bezpiecznika

uzi18, 2017-01-29:
v0.1.0.96
        - dodanie mozliwosci przypisania offsetu do czujnika przykład offset -10.0 deg.:
          OFFSET_T1=-100

    v0.1.0.97
        - pompa CO - dodanie trybu cyklicznego
        - NTP odpytujemy co 24h
        - wstepna wersja testowa pracy RRM z tlem - domyslnie wylaczona

uzi18, 2017-01-30:
v0.1.0.98
    - czesciowe dodanie nowej wersji dopalania
    - podajnik: dodanie informacji w terminalu o czasie wystapienia
      wysokiej temperatury wraz z danymi o jej wartosci
    - podajnik: dodanie wczesnego zabezpieczenia przez wysoka temperatura podajnika
    - w pierwszej fazie tła używamy poprawnej mocy nadmuchu

uzi18, 2017-01-31:
v0.1.0.103
        - dallas: tablice drukujemy przy uruchomieniu tylko jesli znaleziono czujniki
        - dallas: nie drukujemy czasu pobierania danych z czujnikow
        - NTP: komenda N w terminalu pobiera czas z NTP

    v0.1.0.102
        - poprawki poczatkowego odczytu z dallas

    v0.1.0.101
        - opóźnienie uśredniania odczytu

    v0.1.0.100
        - mapowanie: poprawienie wstępnego odczytu

    v0.1.0.99
        - dallas: odrzucamy zbyt duze zmiany pomiaru z czujnika
        - dodanie przeladowania pliku config.txt z www jak i terminala

uzi18, 2017-02-01:
v0.1.0.104
        - dodanie testu wygaszenia kotla - stosowne informacje pojawiaja sie w terminalu
        - MAX31855/6675: usuniecie z wykresu "piły" po restarcie sterownika
        - detektor: poprawienie monitorowania polokresow
        - drobne poprawki bledow kompilatora

uzi18, 2017-02-06:
v0.1.0.105
        - dodanie algorytmu wielomocowego (6szt.) do testów

    v0.1.0.106
        - przechodzimy z tła w podtrzymanie po przebiciu ponad zadana+przebicie

uzi18, 2017-02-06:
v0.1.0.107
    - dokończenie nowego dopalania - gotowe do testów
    - w trakcie pracy tła przekroczenie zadana+max_przebicie wyłącza tło całkowicie
    - tło i dopalanie nie mogą być włączone jednocześnie

uzi18, 2017-02-06:
v0.1.0.108
    - poprawka wczytywania parametrów nadmuchu wielomocowego i kontrola jego wartości

uzi18, 2017-02-07:
v0.1.0.109
        - dodanie statystyk do algorytmu wielu mocy

    v0.1.0.110
        - dodanie debugu do max31855k
        - poprawki w statystykach
    
    - dodanie opisu do nowosci - tlo/dopalanie/wielemocy

uzi18, 2017-02-07:
v0.1.0.111
    - poprawka: wyłączamy tło po zbyt dużym przebiciu
    - poprawka: przełączenie dmuchawy na moc tła jeśli przebiliśmy mocami
    - optymalizacja wyboru mocy w trybie wielomocowym

uzi18, 2017-06-27:
v0.1.0.112
        - dodanie do HTTP API /set do zmian parametrow online

    - esp-link firmware
    - dodanie skryptow uzywajacych esp-link

    v0.1.0.113
        - zmiana zachowania pieca po uruchomieniu - startuje w trybie recznym aby pracowal w auto dodac linijke PIEC_TRYB = AUTO
        - w trybie recznym: dodanie mozliwosci sterowania wyjsciami z www/terminala

    - Domoticz: dodanie polaczenia poprzez esp-link
    - Domoticz: drobna poprawka pobierania informacji z esp-link
    - Nettemp beta support
    - Nettemp,Domoticz: prosty test poprawnosci danych przed wyslaniem

    v0.1.0.114
        - dodanie mozliwosci wlaczenia pomiaru wejsc analogowych przez ANALOG=1
          domyslnie wylaczone, dodanie debugu do analogow.

    - esp-link aktualizacja firmware
    - schemat detektora 0 by Marek Gajos

    v0.1.0.115 - poprawienie zasypu zbiornika, dodano mozliwosc dosypywania.
        - aktualizacja strony www w zakresie zasypu

    v0.1.0.116 - dodanie priorytetu dla CWU - CWU_PRIORYTET = WLACZ

    v0.1.0.117 - uaktywnienie mozliwosci sterowania zaworem 4D z panela LCD
    update www

    v0.1.0.118 - zwiekszenie ilosci mocy RRW do 10szt.
        - dodanie mozliwosci wyboru czujnika oraz histerezy dla CWU2

    v0.1.0.119 - dodanie algorytmu zasypowego: ZASYP_MOC=30 ustawia moc nadmuchu dla trybu zasypowego
    - dodanie do listy algorytmow: ZASYPOWY i WYLACZONY

    v0.1.0.120 - poprawka dla histerezy CWU2 w trybie mieszania

    v0.1.0.121 - RTC_ZIMA=1 wymusza czas zimowy dla NTP

    - dodanie projektu czujnika pokojowego sterowanego z Domoticz, przy okazji regulujacego obnizona temperature w Lucjanie

    Domoticz i Nettemp - mozna wysylac do obu na raz z roznym interwalem

    v0.1.0.122 - dodanie trybu AUTOLATO z kilkoma parametrami i trybami pracy
        - pompy poprawka pracy w momencie gdy t_pieca spadnie ponizej minimalnej
        - drukujemy info o nieudanym pobraniu czasu z NTP
        - drukujemy info o znalezionym EEPROM na terminal
    
    # temperatura zewnetrzna auto przejscia CO w tryb lato
    AUTOLATO_TEMP = 0
    # histereza powrotu CO do trybu zima
    AUTOLATO_HISTEREZA = 0
    # temperatura dla twew auto przejsca CO w tryb lato
    AUTOLATO_TWEW = 0
    # dodatkowe tryb pracy autolato
    AUTOLATO_TRYB = 0
    # poza przejsciem w tryb lato wylaczamy calkowicie piec
    #AUTOLATO_TRYB = WYL_ALGO

    v0.1.0.123
        - w trybie lato dodanie mozliwosci ustawienia odrebnego czasu miedzy podtrzymaniami: RR_PODTRZYMANIE_CZAS_LATO
        - dodanie ograniczenia krotnosci podnoszenia zaru domyslnie wylaczone - PODAJNIK_T_WYSOKA_POWTORKI
        - domyslna histereza dla autolato to 1,5deg

uzi18, 2022-02-25:
```
- v0.1.0.130.35
- dodanie nowych harmonogramow:
  PROG_CO - harmonogram pompy CO
  PROG_CYRK - harmonogram pompy cyrkulacji
  PROG_OB - harmonogram obnizonej temperatury
- poprawki dla Uniwersalnych Termostatów - blad w odczycie parametrow
- Zawór 4D - ZAWOR4D-PRESET=50 - (50%) - pozycja od ktorej po uruchomieniu zaczynamy krecic
- Zawór 4D - ZAWOR4D-CZAS_ZMIANY_KIERUNKU=x - przy zmianie kierunku dodaje czas x
- Zawór 4D - ZAWOR4D-TRYB=POZYCJA - otwiera zawor na pozycje PRESET i nim nie porusza
- Ethernet - poprawki
- detektor - poprawki plus zwiekszenie tolerancji synchronizacji
- analogi - poprawki
- TWS - dodanie maszyny stanow
- LCD_WYGAS=x opcja do wlaczenia wygaszacza po okreslonym czasie
```