uzi18/sterownik #56: zabezpieczenie ślimaka podajnika palnika retortowego przed niskim poziomem paliwa w palniku

bzykbzyk, 2017-01-02:
Pomiar temperatury "fajki" palnika (termopara). Przekroczenie nastawionej temp - zatrzymanie wentylatora i uruchomienie ciągłego podawania paliwa przez nastawiony czas w celu podniesienia żaru do rusztu. 
W pracy automatycznej sterownik powinien docelowo obniżyć obroty wentylatora.
Zabezpieczenie ślimaka przed uszkodzeniem spowodowanym długotrwałym, zbyt niskim poziomem żaru w palniku retortowym - szybkie zużycie, utlenianie, ukręcenie końcówki lub zniszczenie zwojów.

uzi18, 2017-01-04:
Jak mamy zamocowac ten czujnik do fajki?
U mnie w palniku fajka jest zeliwna.

Stan17, 2017-01-04:
No właśnie, problem jest ten sam , jak zamontować ten czujnik.
A sam pomiar mógłby być bardzo przydatny.

uzi18, 2017-01-04:
U mnie moglbym sie zastanowic nad montazem gniazda na czujnik w stalej
retorcie. Aby wiedziec czy jest wystarczajaca ilosc zaru na retorcie.
Dalej pozostaje kwestia jak wyprowadzic przewod na zewnatrz... i pozostawic
palnik nadal szczelnym.

bzykbzyk, 2017-01-05:
> Jest to jakis pomysl, tylko jak mamy zamocowac ten czujnik do fajki?
> U mnie w palniku fajka jest zeliwna.
> 
> @Stan17 @mark3k3 @ravikjab  co tym sadzicie? 
> Bylby to moze pomysl na czujnik żaru ktory kiedys byl planowany.
> Pierwotnie mial byc czujnik w palenisku a raczej blisko niego aby uzyskac 
> informacje o ilosci zaru na paleniski a nie w samym kolanie.
> Teoretycznie czujnik temperatury podajnika juz ma ta funkcje o ktorej piszesz.

Ja przed samymi Świętami miałem awarię ślimaka właśnie przez długotrwałe (kilka dni) palenie bardzo nisko w fajce. Całkowicie wyparowały dwa zwoje, na samym końcu ślimaka. Paliłem przez kilka dni na ruszcie awaryjnym, tragedia, tony wody z komina. Żar nie cofnął się tak daleko, aby zadziałał czujnik temperatury podajnika (ustawiony na 50st). Czujnik zbyt niskiego żaru można zamocować opaską do fajki na zewnątrz. U mnie jest wyczystka, nie powinno być z tym problemu. Zrobiłbym zdjęcie jak zatrzymam piec, ale przez kilka dni mają być teraz wyjątkowe mrozy więc chyba nie bardzo będzie kiedy, smarnąłem coś takiego (zielone-opaska stalowa, czerwone-termopara):

[image]
Ten czujnik pełniłby funkcję jedynie tą co w temacie, oprócz tego powinien być czujnik żaru w koronie palnika, np tak:
http://www.ekspertbudowlany.pl/artykul/id3426,spalanie-pod-czujna-kontrolakotly-sas-z-czujnikiem-zaru

uzi18, 2017-01-06:
> pod-czujna-kontrolakotly-sas-z-czujnikiem-zaru

Ok a jak wyprowadzic przewod czujnika na zewnatrz aby wpiac go w sterownik?

bzykbzyk, 2017-01-06:
Przejście przez palnik w okolicach jak na rysunku (analogicznie przez drugą ściankę - blachę mocującą palnik do kotła)
[image]
wywiercić otwory i przeprowadzić kabel za pomocą przepustów/dławików/dławic kablowych metalowych 
[image]
Usunąć z dławików wszystkie gumowe elementy, uszczelnić przez wypełnienie żaroodpornym silikonem/uszczelniaczem. Kabel do termopary oczywiście ognioodporny silikonowy albo mikowy . Podobnie można przeprowadzić przewody od czujnika żaru.

Stan17, 2017-01-23:
W sumie poważna przeróbka.
W lato tego roku chyba wyjmę cały palnik do przeglądu to się przyjrzę pod kontem takiej przeróbki.

uzi18, 2017-01-24:
Aktualnie jest dodana obsluga konwertera MAX6675K, za jakis czas dodam mozliwosc podlaczenia kilku takich czujnikow.
Więc możesz juz u siebie zamocować i przepuscic termoparę oraz organizować konwertery.

uzi18, 2017-08-31:
@bzykbzyk no i co gotowy na testy?

bzykbzyk, 2017-08-31:
@uzi18, na razie kocioł wygaszony. Powiercone dziury, szpeje prawie kompletne (uwaliłem 6675 - czekam na dostawę) jak będzie komplet zabieram się za uruchomienie.

uzi18, 2017-08-31:
No to fajnie,  sporo sie w sofcie pojawilo i masz co testowac.
Daj cynk to dopisze wsparcie dla kilku czujnikow.
Wtedy jak bedzie Ci dzialac to dodamy obsluge czujnika w zakresie fajki.

uzi18, 2017-09-01:
0.1.0.129.3 - dodanie mozliwosci podlaczenia wielu MAX6675K
    np. T3=MAX6675[48.1] zmapowanie z pinem CS nr 48 i filtrowaniem
    - poprawienie informacji o socketach

bzykbzyk, 2017-11-16:
Zamiast termopary na fajce zamontowałem czujnik żaru w palniku. Co prawda nie do końca w takim miejscu jak chciałem, ale może to wystarczy. Od góry w koronie (jak tu http://www.ekspertbudowlany.pl/artykul/id3426,spalanie-pod-czujna-kontrolakotly-sas-z-czujnikiem-zaru)- niewykonalne bo mam ruszt obrotowy. 
Do konstrukcji wykorzystałem odcinek złomowej rurki stalowej fi10, fi5,5 wewnątrz. Jeden koniec zaspawałem, drugi nagwintowałem wewnątrz M6 i wkręciłem na silikonie 230st deczko zmodyfikowaną sondę termopary. Modyfikacja polegała na lekkim potraktowaniu rurki sondy kombinerkami tak, aby luźno przesuwała się wewnątrz śruby montażowej (rura fi10 ma około 8cm, sonda powinna być wepchnięta głęboko, aż do miejsca spawu). Wprowadziłem to od dołu palnika do kanału nawiewu w palniku, górny otwór, przez komorę wyczystki. Wywierciłem otwór w bocznej ścianie palnika i przykręciłem sondę wkrętem samowiercącym. Chciałem wprowadzić do bocznego  otworu nawiewu, ale czujnik musiałby być wygięty. W tej chwili nie palę, nie mam jak przetestować.
Załączam zdjęcia czujnika i jego montaż. Nie mam zdjęcia przejścia przewodu poprzez dławik poza palnik, jak bedzie trzeba to machnę.
[image]
[image]
[image]

uzi18, 2017-11-16:
> pod-czujna-kontrolakotly-sas-z-czujnikiem-zaru)- niewykonalne bo mam

Najwazniejsze aby oplot i termopara wytrzymywala wysoka temperature

Zerknij na czat:
https://gitter.im/sterownik/Lobby