uzi18/sterownik #57: Poprawienie działania "tła"

Stan17, 2017-01-23:
Dodanie do tła dawek paliwa.
W dalszej kolejności musimy przemyśleć automatyzację działania tła, np w trybie LATO nie ma ono sensu.

uzi18, 2017-01-24:
podczas pracy tla dopalanie nie powinno dzialac

Stan17, 2017-01-27:
Dokładnie, tak samo podtrzymanie.

uzi18, 2017-02-06:
Ponizej nowe opcje, działają tylko jesli ustawiono PIEC_ALGORYTM = RRM2

# 1 - wlaczone tlo, 0 -wylaczone tlo
RRM_TLO_WLACZONE=0
RRM_TLO_MOC=20
RRM_TLO_CZAS_PODANIE=10
# w min.
RRM_TLO_CZAS=15

uzi18, 2017-02-06:
w wersji v0.1.0.106 dodano zabezpieczenie przed nadmiernym wzrostem temperatury i przechodzimy wtedy w podtrzymanie
nowa opcja :

#jesli tlo podbije TPIEC 5deg ponad zadana przechodzimy w podtrzymanie
RRM_MAX_PRZEBICIE_TLO=5

uzi18, 2017-02-06:
v0.1.0.107
    - dokończenie nowego dopalania - gotowe do testów
    - w trakcie pracy tła przekroczenie zadana+max_przebicie wyłącza tło całkowicie
    - tło i dopalanie nie mogą być włączone jednocześnie

closed 2017-02-19 by uzi18