uzi18/sterownik #62: Zegarek temperatura powrotu + Termostat ON

ebyna, 2017-01-28:
można dodać: 
1. 4D ochrona Temperatura powrotu?
http://www.jpeg.cz/images/2017/01/28/1HUFm.png
pin 36 a GND zmiany statusu WWW - Termostat 1  -  ON ?  
http://www.jpeg.cz/images/2017/01/28/PBf4.png

uzi18, 2017-10-30:
@ebyna nadal tych funkcji potrzebujesz?
udało Ci się uruchomić Lucjana?

ebyna, 2017-10-30:
> Komu: uzi18/sterownik <sterownik@noreply.github.com>
> 62)
> "

Dziękuję za zainteresowanie.
Nadal się martwię o połączenie ESP z Lucjanem.
LCD - OK
DS18B20 - OK
ESP - OK konsola, NE IP Lucjan
---------- Původní e-mail ----------

(https://github.com/uzi18/sterownik/issues/62#issuecomment-340480543), or 
mute the thread

"

ebyna, 2017-10-30:
Znowu:
Lan --- tak --- WWW lucjan
ESP8266 (Wifi) --- nie --- WWW lucjan?
---------- Původní e-mail ----------

uzi18, 2017-10-30:
> x8FE4VJSyzTRIks5sxesYgaJpZM4Lwhw3)

Na esp nie działa strona www lucjan.
Esp jest do połączenia z portem szeregowym (komendy i debug) i zdalnej
aktualizacji lucjana.

ebyna, 2017-10-30:
Dziękuję bardzo, zrozumiałem to źle i martwiłem się tym przez miesiąc i trzy
części ESPa.
Dziękuję za odpowiedź.
Czy kiedykolwiek myślisz o Wifi ESP - WWW Lucjan?
---------- Původní e-mail ----------

Datum: 30. 10. 2017 23:16:24
62) 

CongQpUqXAks5sxkntgaJpZM4Lwhw3>

ebyna, 2017-10-30:
I nie rozumiem, jak mogę aktualizować przez ESP

aktualizacja esp-link: wiflash -v 192.168.2.6 user1.bin user2.bin gdzie 
192.168.2.6 to adres ip esp-link

aktualizacja Lucjana do najnowszej wersji: aktualizujpiec_esp wczesniej w 
pliku nalezy ustawic ip esp-link

jeśli będzie kiedykolwiek miał trochę czasu, tak może być za pośrednictwem 
TeamViewer?

882 832 816

279htz

---------- Původní e-mail ----------

uzi18, 2017-10-30:
> auth/AAHOU654Ww2bNq8KIRlaI7
> Elpw6yxvRYO326BGwk-63wks5sxkq0gaJpZM4Lwhw3)

To da się zrobić,  w każdej chwili.
Ale nie mam na to czasu, nie znam znam dobrze www.

A co potrzebujesz na www?