uzi18/sterownik #64: Dodać algorytm wykrycia wygaszenia kotła

Stan17, 2017-02-01:
Parametry
włącz/wyłącz algorytm
minimalna temperatura spalin
minimalna temperatura CO
algorytm
zatrzymanie dmuchawy i podajnika po wykryciu spadku temperatury spalin poniżej "minimalna temperatura spalin" przy jednoczesnym utrzymaniu się temperatury CO powyżej "minimalna temperatura CO"
To wszystko oczywiście  w maszynie stanów żeby przy przedłużającym się postoju i spadku T_CO piec mógł się uruchomić dopiero po skasowaniu alarmu lub resecie sterownika.

uzi18, 2017-02-02:
wstepna testowa wersja bez parametrow w v0.1.0.104, dane pojawiaja sie w terminalu, 
alarm nie jest aktywowany aby nie blokowac pracy sterownika.

uzi18, 2017-10-30:
@Stan17 patrzyles moze jak u Ciebie wykrywa?

uzi18, 2018-10-17:
czekam na info ??
chcialem dodac obsluge ale nie wiem czy to funkcjonuje poprawnie

adamski-68, 2020-02-23:
Proszę o ponowne podjęcie tematu. Mam sugestię , żeby defaultowo wyłączała się dmuchawa a drugie wyjście można było wybierać np: podajnik lub pompa co. Jestem chętny do testowania.

uzi18, 2020-02-23:
@adamski-68 wpierw potwierdzmy ze poprawnie u ciebie wykrywa wygaszenie