uzi18/sterownik #65: dodać algorytm wielomocowy

uzi18, 2017-02-05:
dodano na testy w wersji v0.1.0.105

uzi18, 2017-02-06:
v0.1.0.108
    - poprawka wczytywania parametrów nadmuchu wielomocowego i kontrola jego wartości

mecenas8, 2017-02-18:
Przetestowałem trochę algorytm wielomocowy, mam jedną uwagę.

Wg mnie przydałoby się coś w formie PID, ponieważ teraz, podam parametry jakie są ustawione w tej chwili (zadana temperatura 63 stopnie):

# moc 0 domyslna
RRW_PODANIE0=6
RRW_POSTOJ0= 120
RRW_NADMUCH0=30
# moc 1
RRW_PODANIE1=6
RRW_POSTOJ1= 80
RRW_NADMUCH1=38
RRW_HISTEREZA1=12
# moc 2
RRW_PODANIE2=6
RRW_POSTOJ2= 60
RRW_NADMUCH2=46
RRW_HISTEREZA2=24
# moc 3
RRW_PODANIE3=6
RRW_POSTOJ3= 45
RRW_NADMUCH3=55
RRW_HISTEREZA3=36
# moc 4
RRW_PODANIE4=6
RRW_POSTOJ4= 33
RRW_NADMUCH4=63
RRW_HISTEREZA4=48

Załóżmy teraz, że mamy stosunkowo duży odbiór mocy i algorytm pracuje na mocy nr 3, która mniej więcej wystarcza do podtrzymania temperatury na poziomie 63 stopnie minus histereza ponad 3,6 czyli ok 59 stopni. 
Nigdy więc nie osiągniemy w tym stanie temperatury zadanej 63 stopnie, ponieważ algorytm będzie pracował na mocy nr 3, a stosując PID po pewnym czasie uchyb scałkowałby się (część "I") i przeskoczyłoby na moc nr 4 w celu podbicia temperatury w pobliże zadanej. 

Podobnie mając mały odbiór ciepła algorytm pracuje powiedzmy na mocy nr 3, temperatura zaczyna narastać bardzo szybko, ale algorytm przeskakuje kolejno w moc nr 2, nr 1, nr 0 (po przekroczeniu kolejnych pułapów histerezy), co powoduje przebicie temperatury i wejście w podtrzymanie. Gdyby zastosować PID to część "D" "zauważy" że temperatura szybko narasta i przeskoczy z mocy nr 3 od razu w moc nr 1 bądź nr 0 powodując zmniejszenie narastania temperatury.

uzi18, 2017-02-18:
Takie sa zalozenia algorytmu z histereza i tu sie juz pewnie niewiele
zmieni.
To jest baza.
Nastepny etap to rozne algorytmy wyboru mocy z tych zdefiniowanych, moze
nawet jakis PID albo predzej PD.