uzi18/sterownik #66: Termopara MAX31855K - błędne działanie, zakłócenia z detektora zera

mecenas8, 2017-02-06:
Problem z działaniem termopary na MAX31855K. 

Termopara podaje zaniżone odczyty, zmierzone temperatury wahają się od ok. 5-10 stopni C w stanie nadzoru (czasami spada poniżej zero - wtedy pojawiają się skokowe odczyty od bardzo dużych wartości dodatnich do bardzo dużych ujemnych). Maksymalna zmierzona temperatura podczas pracy kotła to ok 80-90 stopni, gdzie mierząc w tym czasie inną termoparą podłączoną do multimetru rzeczywiste temperatury wynoszą powyżej 150 stopni.

Załączam plik z odczytami z terminala, może w czymś pomoże.

[Sterownik- terminal.txt](https://github.com/uzi18/sterownik/files/756132/Sterownik-.terminal.txt)

Dodatkowo zauważyłem w odczytach z terminala komunikaty: TRIAC: zaklucenia z detektora...
Dmuchawa nie jest "czysto" sterowana, przy mniejszych mocach słychać nierównomierną pracę. 
Sprawdziłem przebieg na pinie 2 Arduino, wygląda jak poniżej, wydaje się być w porządku. Jedyne co zauważyłem to to, że lekko pływa szerokość impulsu, ale to w granicach 1-3%.

[image]

[image]

uzi18, 2017-02-07:
zauwaz ze masz sporo takich komunikatow: TRIAC: zaklucenia z detektora 
poniewaz sygnal z detektora pływa (ok.1ms) to i sterowanie triakiem tez pływa

Przelacz sie z triakiem pod pin 32, na A2 bywały zakłucenia - potrafily przechodzic pelne polokresy.
Sprawdz czemu plywa sygnal z detektora.

Co do termopary - dodaj linijke:
T3 = MAX31855[1]
pod T3 pojdzie bezposredni odczyt z chip-a i wrzuc tutaj wykres

mecenas8, 2017-02-07:
[image]

T3 - odczyt bezpośredni z termopary
tSpaliny - odczyt uśredniony (z dołożonym offsetem +25 st.C

MAX31855K mam z sampli od maxim-ic, powinien być raczej pewny. W razie czego mam zapasowe, mogę przelutować.

Konwerter sam robiłem, są to dzielniki rezystorowe na szynach SCK i CS, SDO póki co idzie bezpośrednio na Arduino, które powinno poprawnie widzieć poziom 3v3 jako High.

Z detektorem to sprawdzę później to przełączenie na 32.

uzi18, 2017-02-07:
Zasilasz maxa z 3v3 jak rozumiem.
Czyli zostaje jeszcze sprawdzic go na jakiejs bibliotece pod arduino ide,
 dasz rade?

mecenas8, 2017-02-07:
Dałbym radę, zajmuję się programowaniem uC prawie na co dzień, z tym że głównie Microchipa. Kwestia znalezienia trochę na to trochę czasu.

Mogę też wpiąć się analizatorem na SPI, wtedy byłoby widać co leci z MAX-a do Arduino.

uzi18, 2017-02-07:
dobra inaczej zaktualizuj do wersji 110:

    v0.1.0.110
        - dodanie debugu do max31855k
        - poprawki w statystykach
    
    - dodanie opisu do nowosci - tlo/dopalanie/wielemocy

jak wcisniesz V, w terminalu co 2s. pojawia sie linijki 
MAX .... z danymi odczytanymi z SPI, danymi przesunietymi w prawo o 18bit (tu ujemnych nie zobaczysz poprawnie) oraz temperatura po konwersji
Mozesz sprobowac sprawdzic analizatorem czy dane zgadzaja sie z wyslanymi na magistrale.
Wklej mi tez log z terminala.

mecenas8, 2017-02-07:
[Sterownik- terminal2.txt](https://github.com/uzi18/sterownik/files/759165/Sterownik-.terminal2.txt)

Na razie nic więcej nie robiłem tylko zaktualizowałem soft i odczytałem z terminala.

mecenas8, 2017-02-07:
Nie rozumiem tego zapisu z terminala.

Przykładowo:
MAX data: 0x1a20, t:184, t:11

Dane z SPI: 0x1A20 - to jest 16 bitów, natomiast MAX wysyła 32 bity.

uzi18, 2017-02-07:
Mniejsza o to ze drukuje tylko 16bit - to poprawie - ale pokazuje ... blad SCV - zwarcie termopary do VCC
mozesz to sprawdzic?

mecenas8, 2017-02-15:
Nie było zwarcia termopary do VCC ani GND. 

Ze względu na brak czasu na dalsze dochodzenia z MAX31855 zakończyłem akcję z termoparą w ten sposób, że przelutowałem scalak na MAX6675 i teraz termopara działa bez zarzutu, podaje poprawne temperatury.

uzi18, 2017-02-19:
Z naszych doswiadczen wynika ze 6675 lepiej toleruje rozne termopary i generalnie lepiej sobie radzi z pomiarem, mimo ze producent sugeruje uzycie 31855

uzi18, 2017-03-20:
Generalnie chcialem podsumowac, jest to raczej problem z samym MAX31855 lub termopara - bo uklad nie wspolpracuje ze wszystkimi rodzajami termopar K.

uzi18, 2017-08-31:
Czyli lepiej jednak bawic sie z MAX6675K, zamykam

closed 2017-08-31 by uzi18