uzi18/sterownik #70: EEPROM problemz wykrywaniem

mipurel, 2017-08-13:
Mam moduł RTC na DS3231 + AT24C32, Lucjan w wersji v.0.1.0.126 rtc wykrywa normalnie natomiast nie wykrywał pamięci. Problemem jest sprawdzanie adresu przez Lucjana ponieważ moduł domyślnie EEPROM ma ustawiony pod 0x57, przy zmianie adresu na 0x50 został wykryty. Wniosek aby do Lucjana dodać zakres adresów od 0x50 do 0x57 ponieważ zgodnie ze schematem adresowania 1:0:1:0:A2:A1:A0 pokryje to wszystkie możliwość lub chociaż domyślny dla modułu. @uzi18 w związku z pracami nad zapisem konfiguracji do EEPROM planujesz to zapisywać w ATmedze czy zewnętrznej pamięci na I2C?

uzi18, 2017-08-13:
Nie wiem czy jest sens szukania pod innymi adresami,  chyba ze znajdzie sie
sensowny argument, to moge to rozwazyc.

Mysle ze w zew.  eeprom bedzie konfig ale to wyjdzie podczas testow.
Poniewaz mam kilka pomyslow jeszcze,
wiec nic nie jest przesadzone.

mipurel, 2017-08-13:
Argument to był tylko taki, że 0x57 jest domyślny w tym konkretnym module - na 0x50 trzeba pozwierać wejścia adresowe, trochę dochodzenia było czemu nie jest wykrywany.

uzi18, 2017-08-13:
Z tego co wiem to akurat ds3231 nam sie nie sprawdzi,  bo nie ma
dodatkowego ramu, ktory potrzebujemy.

Poza tym moduly ds1307 i ds3231 nie sa zgodne.

Reasumujac nie uzywamy tego modulu,  kupowanie go dla samego eepromu mija
sie z celem.

O dodaniu testu dla eeprom pod adresem 0x57 postaram sie pamietac.

mipurel, 2017-08-13:
Aktualnie ds3231 jako RTC do Lucjana się sprawdza, jest pod tym samym adresem co ds1307, którego też mam ale bez eepromu. Potrzeba dodatkowego RAMu wydaje mi się jednak priorytetowa i z nowym modułem nie powinno być kłopotu.

uzi18, 2017-08-13:
Ram w rtc uzywamy do zapisania stanu zasobnika,  aby nie zajechac eepromu.

uzi18, 2017-08-13:
W wersji 0.1.0.127 dodalem poszukiwanie adresu 0x57

14.08.2017 12:00 AM "Bartłomiej Zimoń" <cactus.list@gmail.com> napisał(a):

Ram w rtc uzywamy do zapisania stanu zasobnika,  aby nie zajechac eepromu.

uzi18, 2017-08-13:
Wrzuciłem na github mozesz przetestowac.

uzi18, 2017-08-14:
dopisalem jest w kodzie od wersji 0.1.0.127, prosze o info z testow

mipurel, 2017-08-14:
Wykrywanie EEPROM pod adresem 0x57 działa.

closed 2017-08-25 by mipurel