uzi18/sterownik #71: Sterowanie dmuchawą AC - sygnał odwrócony

mipurel, 2017-08-13:
Wprowadzenie sterowania dmuchawą AC dla modułów z triakami wyzwalanymi niskim poziomem (SSR low level trigger). Sygnały na 32 i A13 są identyczne, można np. jeden z nich odwrócić lub wykorzystać dodatkowy pin albo zrobić to programowo w pliku config.

uzi18, 2017-08-13:
Prawie wszystkie dostepne ssr-y maja detekcje przejscia przez zero co
 powoduje,  ze nie da sie fazowo sterowac moca dmuchawy.
Nie spotkalem sie jak dotad z modulem,  ktory nadawałby się do tej funkcji.

Sam sie przekonaj,  dodaj inwerter sygnalu i wpusc na modul ssr, nastepnie
zmieniaj nadmuch w trybie recznym.

Sprawdz dokumentacje samego przekaznika,  jaki masz na nim nadruk?

mipurel, 2017-08-14:
Ok, ale ja mam przelutowanego omrona G3MB-202PL w jednym kanale z przelaczaniem w dowolnej fazie. Dodatkowy inwerter nieco komplikuje układ a miało być jak najprościej, a najlatwiej jest zmodyfikowac program. Może moje uwagi  w temacie są błache, ale spodobał mi się twój projekt i chciałbym włączyć aktywnie w rozwój. Troszkę (może trochę więcej) znam się na elektronice chociaż miałem dłuższą przerwę w tej dziedzinie i ogólnie interesuję się zagadnieniami automatyki inteligentnych domów. Jak chcesz mogę ci podesłać do testów.

uzi18, 2017-08-14:
To zmienia postac rzeczy i w sumie od tego mogles zaczac :)
Pamietaj ze staramy się  używać  modułów  ogolnie dostepnych, stad taka a
nie inna odpowiedz.

Musisz miec na uwadze aby wyjscie pinu podciagnac do +5v, a dmuchawe
dodatkowo odcinac przekaznikiem.
Musi byc pewnosc ze dmuchawa nie uruchamia się gdy procesor jest w trybie
reset/bootloadera i programowania.
Tak samo w razie przegrzania kotla/podajnika termik powinien rozlaczyc
przekaznik.

Czemu nie zalatwiles sobie modulu high trigger?

uzi18, 2017-08-14:
v0.1.0.128 - dodanie sygnalu odwroconego dla dmuchawy AC na pin 33

prosze o info z testow, w pdf od przekaznika jest info ze max czas zalaczania to 
1ms ... ciekawe ile jest typowo, moze sie okazac ze rozwiazanie sie nie nadaje.

mipurel, 2017-08-14:
Z modułem wyszlo w trakcie, trochę też nie zauważyłem, że przerzuciłeś z A13 na 32 bez odwrócenia sygnału tak jak pozostałe. Poza tym te low level 4ch były na Ali cost effective. Sprawdzę ale w środku tego SSR jest tylko optotriak i triak ze snubberem, robiłem symulacje dla takiego modułu sterowanego cyfrowo w proteusie. Przetestujemy zobaczymy. [https://www.google.pl/imgres?imgurl=https%3A%2F%2Fwww.ia.omron.com%2Fsupport%2Fguide%2F18%2Fimg%2FSSR_TG_Overview_fig2.jpg&imgrefurl=https%3A%2F%2Fwww.ia.omron.com%2Fsupport%2Fguide%2F18%2Foverview.html&docid=FzqbVtBm1E6JRM&tbnid=XAx7Dptw4T9cTM%3A&vet=10ahUKEwjQ0c7CwtbVAhVlSZoKHeiyCdcQMwhdKBUwFQ..i&w=357&h=426&hl=pl-PL&bih=564&biw=360&q=omron%20solid%20state%20relay&ved=0ahUKEwjQ0c7CwtbVAhVlSZoKHeiyCdcQMwhdKBUwFQ&iact=mrc&uact=8](url)

uzi18, 2017-08-14:
> omron.com%2Fsupport%2Fguide%2F18%2Fimg%2FSSR_TG_Overview_
> fig2.jpg&imgrefurl=https%3A%2F%2Fwww.ia.omron.com%2Fsupport%2Fguide%2F18%
> 2Foverview.html&docid=FzqbVtBm1E6JRM&tbnid=XAx7Dptw4T9cTM%3A&vet=
> 10ahUKEwjQ0c7CwtbVAhVlSZoKHeiyCdcQMwhdKBUwFQ..i&w=357&h=426&
> hl=pl-PL&bih=564&biw=360&q=omron%20solid%20state%20relay&ved=
> 0ahUKEwjQ0c7CwtbVAhVlSZoKHeiyCdcQMwhdKBUwFQ&iact=mrc&uact=8 <http://url>

aktualne podlaczenia zawsze sa w pliku polaczenia.txt

Skopiowalem wyjscie na 32 z tego powodu ze wyjscie na porcie z analogowymi
wejsciami przepuszcza czasem caly polokres.
Na pinie 32 tego efektu nie bylo.

mipurel, 2017-08-14:
Wejścia modułu SSR low level trigger są podciągnięte do plusa  co powoduje, że bez wymuszonego stanu niskiego są wyłączone. Na chwile obecną tylko sprawdziłem oscyloskopem i widzę, że zmieniłeś wyzwalanie na impulsowe dzięki czemu wyzwolenie triaka powinno być pewniejsze. Wcześniej po jasności diody można było ocenić moc dmuchawy ale to szczegół :). Mam ten sterownik w stanie stołowo-testowym więc trochę muszę pokombinować z obciążeniem. Jak tylko będę po testach dam znać.

uzi18, 2017-08-14:
Masz 2 typy sterowania DM_TYP=0 lub 1
Domyslnie typ jest stary czyli 0.
Na nowym typie sterowanie bramka jest impulsem.
Na starym wylaczalem w zerze.

Wyglada ze oba testowales na oscyloskopie?

mipurel, 2017-08-15:
Testowałem v.0.1.0.126 i 128. Wychodzi na to, że w 126 zmiana parametru DM_TYP=1 nic nie daje przebieg jest jak dla starego sterowania na wyjściu 32. W 128 nie wiem czy efekt był zamierzony jest impuls ok. 300us, i seria podwójnych szpilek do końca półokresu zaczynają się z rożnym opóźnieniem w stosunku do głównego impulsu odstęp między nimi ok 500us. Zasadniczo stosuje się dodatkowe impulsy i pierwotnie sądziłem, że taki był zamiar ale po twojej odpowiedzi nie jestem już pewien. Jeżeli triak jest włączony i tak nie będą generowały dodatkowych zakłóceń.

uzi18, 2017-08-15:
Wiem ze dodatkowe impulsy sie stosuje ale to co piszesz to nie jest
działanie zamierzone.
Mozesz wrzucic tu oscylogramy?

uzi18, 2017-08-15:
Wklej mi co Ci sie pojawia w terminalu po wyslaniu cyfry 3

mipurel, 2017-08-15:
TRIAC: T-9948 Z-0 T-Z-604 F-0 50% DET-221/200
TRIAC: 0-93 1-6 S-99 ok-200
TRIAC: T-9948 Z-0 T-Z-604 F-0 50% DET-221/200
TRIAC: 0-93 1-6 S-99 ok-200
TRIAC: T-10048 Z-0 T-Z-504 F-0 50% DET-221/200
TRIAC: 0-93 1-7 S-98 ok-200
TRIAC: T-9948 Z-0 T-Z-604 F-0 50% DET-221/200
TRIAC: 0-92 1-7 S-98 ok-200
TRIAC: T-9868 Z-0 T-Z-604 F-0 50% DET-221/200
TRIAC: 0-92 1-7 S-98 ok-200
TRIAC: T-9852 Z-0 T-Z-704 F-0 50% DET-222/200
TRIAC: 0-92 1-7 S-98 ok-200
TRIAC: T-9952 Z-0 T-Z-600 F-0 50% DET-221/200
TRIAC: 0-93 1-6 S-99 ok-200
TRIAC: T-9948 Z-0 T-Z-604 F-0 50% DET-221/200
TRIAC: 0-93 1-6 S-99 ok-200
TRIAC: T-9952 Z-0 T-Z-600 F-0 50% DET-221/200
TRIAC: 0-93 1-6 S-99 ok-200
TRIAC: T-9848 Z-0 T-Z-704 F-0 50% DET-221/200
TRIAC: 0-92 1-7 S-98 ok-200
TRIAC: T-9948 Z-0 T-Z-604 F-0 50% DET-221/200
TRIAC: 0-93 1-6 S-99 ok-200
TRIAC: T-9948 Z-0 T-Z-704 F-0 50% DET-221/200
TRIAC: 0-92 1-7 S-99 ok-200
TRIAC: T-9948 Z-0 T-Z-604 F-0 50% DET-221/200
TRIAC: 0-93 1-6 S-99 ok-200
TRIAC: T-9948 Z-0 T-Z-604 F-0 50% DET-221/200
TRIAC: 0-93 1-6 S-99 ok-200
TRIAC: T-9952 Z-0 T-Z-704 F-0 50% DET-221/200
TRIAC: 0-92 1-7 S-99 ok-200
TRIAC: T-9952 Z-0 T-Z-700 F-0 50% DET-221/200
TRIAC: 0-92 1-7 S-99 ok-200
TRIAC: T-10052 Z-0 T-Z-500 F-0 50% DET-221/200
TRIAC: 0-93 1-7 S-98 ok-200
TRIAC: T-9952 Z-0 T-Z-600 F-0 50% DET-221/200
TRIAC: 0-94 1-5 S-100 ok-200
TRIAC: T-9948 Z-0 T-Z-604 F-0 50% DET-221/200
TRIAC: 0-93 1-6 S-99 ok-200
TRIAC: T-9948 Z-0 T-Z-604 F-0 50% DET-221/200
TRIAC: 0-93 1-7 S-99 ok-200
TRIAC: T-9948 Z-0 T-Z-604 F-0 50% DET-221/200
TRIAC: 0-93 1-7 S-99 ok-200
TRIAC: T-9948 Z-0 T-Z-604 F-0 50% DET-221/200
TRIAC: 0-93 1-6 S-99 ok-200
TRIAC: T-9948 Z-0 T-Z-604 F-0 50% DET-221/200
TRIAC: 0-93 1-6 S-99 ok-200
TRIAC: T-9948 Z-0 T-Z-704 F-0 50% DET-222/200
TRIAC: 0-92 1-7 S-99 ok-200
TRIAC: T-9948 Z-0 T-Z-604 F-0 50% DET-221/200
TRIAC: 0-93 1-6 S-99 ok-200

uzi18, 2017-08-15:
Pieknie detektor dziala :) napisalem Ci email na google

uzi18, 2017-08-15:
Jaki detektor zmontowales, masz bardzo stabiliny sygnal :)

mipurel, 2017-08-25:
Detektor jest podobny do tego przesłanego przez Marka Gajosa, 22k/2W x3 + PC817 + LED + Mostek + zener do zabezpieczenia.

closed 2017-08-25 by mipurel