uzi18/sterownik #72: WWW: dodać dostęp do config.txt na hasło

mipurel, 2017-08-31:
Ze względu na to, że sposób dostępu jest inny niż w  #20  oraz jedno nie wyklucza drugiego robię to w osobnym issue.
-dodanie hasła dla prostego zabezpieczenia configu ja to widzę tak, że na www jest pole tekstowe do wpisania hasła przy wciśnięciu odczyt lub zapis jest przesyłane jest hasło i porównywane z zapisanym w procesorze w przypadku zgodności następuje odczyt lub zapis tak jak to jest obecnie w przypadku braku zgodności w pole tekstowe gdzie wyświetla config jest wyświetlana stosowna informacja o braku uprawnień
-hasło zapisywane w ATmedze np. w eepromie z poziomu terminala szeregowego coś w stylu “sHASLO=NOWEHASLO STAREHASLO

uzi18, 2017-08-31:
to moze byc ciekawy pomysl

uzi18, 2017-09-24:
dodano ograniczenie dostepu do strony na haslo