uzi18/sterownik #74: Rozszerzenie algorytmu ZASYPOWY

mipurel, 2017-08-31:
Parametry algorytmu zasypowego proponuję zrobić analogicznie do RR (albo RRM, RRM2) tylko bez podajnika, czas podtrzymania w minutach, reszta w sekundach:
# gdy=0 sterownik pracuje tak jak teraz bez przedmuchów (w RR minimalna wartość to 1 )
ZASYPOWY_PODTRZYMANIE_CZAS = 10
ZASYPOWY_PODTRZYMANIE_NADMUCH = 10
ZASYPOWY_PODTRZYMANIE_MOC = 30
# gdy > 0 i sterownik jest w trybie lato korzystamy z czasu lato (niekonieczne bo zasypowy i tak wymaga dużo większej obsługi - ale fajne):
ZASYPOWY_PODTRZYMANIE_CZAS_LATO = 0

ZASYPOWY_PRACA_NADMUCH = 10
#zamiast ZASYPOWY_MOC
ZASYPOWY_PRACA_MOC = 40

Ewentualnie z globalnym przełączeniem w tryb ZASYPOWY i wyłączeniem podajnika - pozostałe parametry jak dla trybu RRM2.

UWAGA: RRM_PODTRZYMANIE_CZAS = 0 nie wyłącza podtrzymania prośba o skorygowanie działania

uzi18, 2017-09-26:
gdy = 0 - co ma byc = 0?

mipurel, 2017-09-26:
Gdy  ZASYPOWY_PODTRZYMANIE_CZAS = 0 wtedy  brak podtrzymania(przedmuchów). Aktualnie  w algorytmie RR(M,M2) nie możną wyłączyć podtrzymania poprzez wpisanie RR_PODTRZYMANIE_CZAS=0 zostawia wtedy ostatnią zapisaną wartość dlatego sygnalizuję żeby to poprawić.

uzi18, 2017-09-27:
Postaram sie poprawic podtrzymanie w algorytmach aby wpisanie 0 spowodowalo
jego omijanie.

uzi18, 2018-10-17:
nadal potrzebne?