uzi18/sterownik #75: Dodanie przejścia w stan podtrzymania

kamilb9, 2017-09-21:
Przydałoby się do niektórych instalacji bez zaworu trójdrożnego i z działającą grawitacją dodać możliwość przejścia w stan podtrzymania żaru w piecu po osiągnięciu przez TWEW odpowiedniej temperatury. Np. u mnie nie wystarczy wyłączyć pompkę lub przejście w cykliczne jej działanie bo działa nadal obieg grawitacyjny piec cały czas dogrzewa dom  -  a jak wiadomo większa temperatura w domu to większe straty - gdy niema odpowiedniego docieplenia. Przejście w stan podtrzymania po osiągnięciu w domu odpowiedniej temperatury z ustawioną histerezą byłby przydatny ponieważ nie trzeba by rozpalać ponownie.

uzi18, 2017-09-24:
w 0.1.0.129.6 dodano obnizenie zadanej na piecu w momencie przekroczenia zadanej dla twew

uzi18, 2017-09-24:
polecam zamontowac sobie zawor 4D lub termostaty na kaloryferach

uzi18, 2020-04-06:
@kamilb9 sprawdziles czy ci to dziala?