uzi18/sterownik #77: układ przeciw blokowaniu pomp.

sargus123, 2017-10-28:
mam propozycję aby dodać układ przeciw blokowaniu pomp raz na dobę odpalić pompy na 30 sekund. przeciw blokowaniu. jak cały sezon stoją to różnie bywa.

uzi18, 2017-10-28:
@Stan17 @mark3k3 co o tym sadzicie?

Tzn maja sie uruchamiac gdy jest lato czy caly rok?

sargus123, 2017-10-28:
łatwiej jest to puścić przez prostą komendę niż konfigurować lato zima. 
wystarczy \ze ma włączyć pompy np codziennie o godz 8.00 a w config puścić komendę gdzie wpisujemy wybrane pompy do zruszenia ich  np. 
ANTY-STOP pompa: CO , CWU , CUW2 itd..
albo  1/0 on of aktywacji dobowej. 
ANTY-STOP-CO =1
ANTY-STOP-CWU=0
ANTY-STOP-CWU2=1
itd

Poco komplikować dodatkowo ustawieniami zimą tak naprawdę powinno załączać strefy gdzie ktoś ma wyłączone bo nie chce grzać (np. podłogówka w nieużywanym pokou/budynku). 
tak naprawdę do zimnych instalacji w niektórych sterownikach jest to też system anty zamarznięciu ale to wtedy dłużej pompami kręci.

uzi18, 2017-10-28:
Pytam bo ktos zglaszal ze chcial cos takiego w lato raz w tygodniu puszczac

sargus123, 2017-10-28:
Raz w tygodniu też może być. 
Kwestia tego co mniej kodu zabierze.
 no dziś jest miejsce a jutro... będzie już brakować. :)

uzi18, 2017-10-28:
Na jutro to szykuje sie Lucjan 2.0 i tam miejsca jest duuuzo wiecej.

Stan17, 2017-10-28:
Temat był już kiedyś poruszany i został odłożony na później.
Przydatne do pomp starego typu, nowe elektroniczne robią to same.
Interwał załączenia raz na tydzień jest w zupełności wystarczający, i żaden znaczący sposób nie zakłóci pracy instalacji więc rozdzielanie na zima i lato nie jest konieczne.

uzi18, 2017-10-28:
Ok wiec mogloby to wejsc,  na 1raz w tygodniu np. na 1min.

Stan17, 2017-10-28:
No dokładnie.

sargus123, 2017-10-28:
''Przydatne do pomp starego typu, nowe elektroniczne robią to same. ''- nie zrobią jeśli nie mają podanego napięcia . dlatego to raz na tydzień jest dobre.

Stan17, 2018-02-08:
I co z tematem ? Właśnie dzisiaj odkryłem że cyrkulacyjna jest zablokowana.

uzi18, 2018-02-08:
Wszystko sie da :)