uzi18/sterownik #78: sterowanie 2 okładami centralnego ogrzewania.

sargus123, 2017-10-28:
Mam instalację rozbudowaną na kilka stref centralnego ogrzewania. jedna pompa chodzi cały rok a inne są włączane gdy temperatura spadnie poniżej  13 stopni. 

może dopisać warunek  z ustawieniem w configu. 
 jeśli tZEW jest poniżej [ ...] to  załącz pompę [ podajemy jaką ]

uzi18, 2017-10-28:
ta pompa pracujaca caly rok to pompa CO? Moze warto dodac pompe mieszajaca, 
pod ktora moglbys ja podlaczyc a tamta 2ga wpiac jako CO w trybie AUTOLATO

sargus123, 2017-10-28:
Tak pierwsza pompa to pompa transmisyjna tłoczy wodę w głównej magistrali. do bojlera (przelotowo bezpośrednio przelatuje), budynku A,B,C i do bojlera głównego w budynku C
Do niej są podłączone budynki poprzez wymienniki płytowe gdzie w takim połączeniu musi być  pompa.  (mieszacza nie mogę podłączyć różnica ciśnień i część układu przechodzi w tryb chłodzenia /coś ala woda lodowa/  mieszacz jest w części budynku A 

Wiem że nie obsłuży mi wszystkiego Lucjan z powodu odległości. a magistrale mam tylko internetową.

uzi18, 2017-10-28:
Jesli siec masz wspolna w budynkach to posrednio da sie to ugryzc.

sargus123, 2017-10-28:
tak mam sieć wspólną w budynkach.  :)

uzi18, 2017-10-28:
Jesli jest to wszystko da sie zrobic

sargus123, 2017-10-28:
to jestem gotów na propozycje  :) i z pewnością nie jednemu się przyda. kto ma instalację rozproszoną.
jeśli chodzi o obsłużenie wszystkich elektro zaworów, pomp, i zaworów 3 drożnych. Pomijając awaryjną linię zasilania w postaci pieca gazowego co przejmuję kontrolę grzania jak padnie piec na groszek. 
muszę znaleźć program do rysowania projektów kotłowni by rozpisać wizualnie instalację wtedy można zobaczyć jak to razem można sprzężyc we wspólne działanie.

uzi18, 2017-10-28:
Do wszystkich prawie informacji jest dostep przez api www/serial,  wystarczy z nich skorzystac.
Np.  Wrzucic  je sobie do domoticz i nim przekazac dalej do espeasy czy czegokolwiek.

uzi18, 2017-10-28:
Wystarczy Ci pompa cwu cyrkulacji jako podstawowy obwod czy musze dodac nowa do mieszania?

uzi18, 2018-10-17:
update??

closed 2018-10-20 by uzi18