uzi18/sterownik #79: wykożystanie termopary do sprawdzenia czy jest używane 2 palenisko  i ew. czy jest dosypać paliwa w podajniku.

sargus123, 2017-10-28:
czy termopara {tp} może wykrywać 'anomalie' gdy w piecu ktoś dołoży drewno/papier na ruszt aby wyłączyła podajnik ślimakowy a nadal podawał powietrze. ( np. po wzroście temperatury na {tp} ponad normę skraca czas podawania paliwa na ślimaku o np. 70% a wentylator dalej wg. ustawień z RRM2.
jeśli jest dłuższy spadek temperatury na {tp} poniżej normy to zwiększa podawanie paliwa na podajnik np. o 30% 
w configu można podać w ustawieniach np pozycję do każdej mocy w  RRM2  
TERMOPARA-TEMP.MIN:  
TERMOPARA-TEMP.MAX:  
gdzie podajemy zakres temperatur w normalnej pracy. 
jeśli będą rozbieżności poza zakresy to odpowiednio zachowa się  sterownik względem podajnika. na danej mocy kotła. (nie wiem tylko czy trzeba dodać zwłokę czasową po zmianie mocy kotła uruchomię jeszcze i dokładnie sprawdzę wahania.)

sargus123, 2017-10-28:
tak się zastanawiam że pokrewnie może też to przeciwdziałać tym co opisane w: 
*56  zabezpieczenie ślimaka podajnika palnika retortowego przed niskim poziomem paliwa w palniku #56

gdy jest za nisko to materiał się pali gorzej i daje niższą temperaturę spalin. 
jeszcze zastanawiam się czy  dodając jeszcze czas do powrotu do odpowiedniej temperatury spalin można określić czy jest wszystko ok z podawaniem paliwa ( czy się nie zawiesiło, zerwało ślimaka, brak paliwa) ale to w wykresach będzie wtedy to przestudiować.

uzi18, 2017-10-28:
Tak wykresy bardzo sie przydaja.

sargus123, 2017-10-30:
Przeglądałem wykresy i zauważyłem przy moim piecu  ( w mniejszej mocy czy sprawności może być inaczej) temperatura jako wysoka to 250 stopni jeśli jest wrzucone paliwo na ruszt to temperatura wzrasta nawet do 400 stopni. 
 podobnie jeśli chodzi o dolną granicę to dolny próg w mocy wychodził na 200 stopni. 

Z tego co widzę to jest możliwość na podstawie tego wykorzystać do sterowania. W moim przypadku jeśli temperatura będzie powyżej 330 stopni (30 stopni zapasu ) to podajnika  czas zmniejszyć o  np 50% i wentylator też obniżyć o 40%  ( poco tyle ciepła miało by w komin zwiać. z cennego materiału opałowego)

i miałem sytuację zawieszenia ( włączyła się elektroniczna zawleczka ) i temperatura zaczęła spadać. więc też to widać na wykresie. 

Moja propozycja zapisu w config do programu . no a  w kodzie to wiadomo.. :) 

*wykrywanie opału na ruszcie awaryjnym  0/1 - wł /wył
RUSZT-AWARYJNY= 1
TERMOPARA-TEMP.MAX: 330
*moc wentylatora o ile odjąć jej w %
RUSZT-WENT = 40
*czas podawania ile czasu dodać w sekundach pomiędzy podawaniem paliwa
RUSZT-PODAJNIK = 5

*wykrywanie braku paliwa w palniku 0/1 - wł /wył
PALNIK-POZIOM= 1
TERMOPARA-TEMP.MIN:  180
*czas podawania ile czasu odjąć w sekundach pomiędzy podawaniem paliwa
PALNIK-PODAJNIK = 5

uzi18, 2017-10-30:
wrzuc tu wykres

sargus123, 2017-10-30:
dziś wrzucam przy pracy normalnej. jutro będę miał po załadowaniu rusztu.
[image]

uzi18, 2017-10-30:
zaktualizuj lucjana,  dzis wrzucilem nowa wersje

sargus123, 2017-10-31:
[image]

adamski-68, 2020-02-23:
Dopisuję się z prośbą o redukcję obrotów dmuchawy przy dochodzeniu do max. TSPALIN. Myślę, że  można by to zrobić dodając dwa parametry do dmuchawy: DM MIN. i DM MAX., wtedy dwuchawa pracuje na maksymalnych zadanych obrotach do momentu zbliżenia się do zadanej max. temperatury spalin kiedy to rozpocznie się redukcja obrotów aż do obrotów minimalnych ustawionych w parametrach. Spadek temp. spalin powoduje wzrost obrotów dmuchawy aż do maksymalnych. Mogłoby to chyba być też być zrobione prościej osiągnięcie max. temperatury spalin przestawia dmuchawę w tryb obrotów minimalnych a spadek temp .spalin o histerezę ( myślę, że tu musiałoby być z 10*C może ustawione systemowo- ze względu na częstotliwośc przełączania) powoduje znów pracę na maksymalnych obrotach. Można też podejść do tego z innej strony a mianowicie utrzymywanie temp. spalin podobnie jak temp. pieca tzn. praca dmuchawy tak by utrzymać stałą temp. spalin ale to chyba nie ma sensu.

uzi18, 2020-03-28:
mysle ze powinno nam sie udac wykrywac palenie na ruszcie