uzi18/sterownik #80: klawisz wywołujący zwloke w dalszej pracy na czas wyczyszczenia pieca

sargus123, 2017-10-28:
Przydałby się guzik dla osób co nie cierpią elektroniki. 
Miałby zawiesić pracę pieca na ok. 15 min aby go wyczyścić. często zdarzało mi się że robione to było za pomocą wyłączenia z sieci ale potem zapominano aby ponownie włączyć piec. a tak sam przystąpi do pracy. 

i ew. drugi co napełniał ślimak paliwem po zawieszeniu albo braku w zasobniku. naciśnięcie dawało by czas podania ok. 2 minut.

uzi18, 2017-10-28:
@Stan17  @bzykbzyk @mark3k3 co o tym myslicie?

Stan17, 2017-10-28:
Zdaje mi się że rozmawialiśmy o tym , i chcieliśmy do tego celu użyć długiego przyciśnięcia pokrętła impulsatora.
Uważam że funkcja przydatna.

uzi18, 2017-10-28:
Dlugie przycisniecie powoduje wejscie w menu trybu ręcznego,  a koledze chodzi o konkretne dodatkowe dzialanie.

Stan17, 2017-10-28:
A nie da rady zrobić jeszcze dłuższego ? Albo dwóch krótkich szybko po sobie ?

sargus123, 2017-10-28:
zgadza się chodzi o dodatkowe bo osobie co ma 70 na karku trudno nieraz wyjaśnić że pokrętło się kręci i ..........  prędzej załapie że naciśnie guzik . Wtedy na ekran można wrzucić komunikat  czyszczenie pieca i czas odliczany wstecz..  i majster będzie wiedział że już może otworzyć piec . 
a jak ma kręcić itp. to się wycofa.  

to już nieraz przerabiam jak uczę osoby obsługi tabletu to nieraz po roku jeszcze nie mogą się połapać czemu ikonki przemieszczają się na ekranie  jak wyciera ściereczką ekran na włączonym tablecie :)

sargus123, 2017-10-28:
problem jest taki że młodzi mają siłę się uczyć a starsi już nie zawsze.

uzi18, 2017-10-28:
2 krotkie odpadaja,  ale z tym bardzo dlugim np. 2s,  moze cos sie da wymyslec

uzi18, 2017-10-30:
3s przytrzymanie byloby ok?
1,2s przenosi do menu recznego... 
Bedzie ktos mogl przetestowac w razie czego?