uzi18/sterownik #81: Lcd 4x20 wyświetlanie stanów wyjść pomp itp.

sargus123, 2017-10-28:
W sterownikach często wyświetlane lub dioda mówi o stanie pracy wyjść . 
Praca pompy CO
Praca podajnika
Praca wentylatora. itd. 

w dolnej linijce można zrobić strzałki w dół w co 2 pozycji wyświetlające się w zależności od aktualnego stanu wyjść. 
np
 2 poz strzałka w świeci kiedy pompa CO pracuje 
 4 poz strzałka w świeci kiedy pompa CWU pracuje 
 6 poz strzałka w świeci kiedy pompa PODAJNIK pracuje 
 8 poz strzałka w świeci kiedy pompa WENTYLATOR pracuje  tu może na podstawie mocy linijki by się wypełniały w zależności od mocy 1 kreska 10% 2kreski 20% 3 kreski 40% ...... 8 kresek 100%
 10 poz strzałka w świeci kiedy 4 drogowy zamyka. 
 12 poz strzałka w świeci kiedy 4 drogowy otwiera.

Wiem że niektórzy wstawiają diody w obudowę ale tak jest czyściej i ładniej.

uzi18, 2017-10-28:
Zerkne na to

uzi18, 2017-10-30:
W dolnej linijce na jakim ekranie proponujesz?
Na duzym wyswietlaczu mozna uzyc dolne 2 linijki do takich informacji
@Stan17 @bzykbzyk  @ravikjab @mark3k3 jak myslicie?