uzi18/sterownik #82: Zachowanie sterownika po zaniku zasilania (wyłączeniu)

Amozeja, 2017-10-29:
Nie znalazłem takiego tematu, więc proponuję: 
Po włączeniu zasilania sterownik w zależności od konfigu włącza się  na pracę ręczną lub automat. I tu może być problem z położeniem zaworu czterodrogowego - skąd sterownik ma wiedzieć, jaki jest rzeczywisty kąt otwarcia zaworu (choćby przez ręczne jego ustawienie i powrót na automat, lub ostatnie położenie przed wyłączeniem, zanikiem zasilania). Czy niemożna byłoby założyć takiego działania:
Sterownik uruchamia się, następuje włączenie pomp co, podłogówki i cwu, podaje komendę do pełnego otwarcia zaworu (np. 120 lub 140 sek - w zależności od siłownika), następnie przez ok. 60 sek działają wszystkie pompy, zawór na max. W ten sposób jest pewne, z jakiego położenia startuje siłownik - na pewno będzie w pełni otwarty i ewentualnie schłodzi trochę podgrzany bez nadzoru piec (w czasie braku zasilania) . Następnie sterownik przechodzi w tryb ustawiony w konfigu. 
Bo na chwilę obecną zauważyłem, że jest możliwość ustawienia wybranego otwarcia zaworu przy starcie, ale nigdy nie można być pewnym, w jakim stanie zawór znajdował się w momencie zaniku zasilania.

uzi18, 2017-10-29:
@ravikjab @mark3k3 @Stan17 @bzykbzyk  rozwazalismy ten temat?

Z tego co ja pamietam to sterownik po uruchomieniu zamyka zawor - wiec stan jest znany
Musze zerknac w kod, chyba ze ktos wczesniej sprawdzi.

sargus123, 2017-10-29:
jeśli niema karty w slocie to sterownik zamyka zawór. 
jeśli karta jest wsadzona  a config niema nic o zaworze to go nie zamyka (ja nie używam teraz zaworu i nic nie wpisywałem odnośnie jego)

uzi18, 2017-10-30:
Wklej mi tu config.txt na ktorym nie zamyka

sargus123, 2017-10-30:
#Konfiguracja sieci:
#Adres IP (dla automatycznej konfiguracji sieci piszemy: NET_IP=auto):
NET_IP=auto
#Adres MAC (w hex):
NET_MAC=1A1B1C1D1E1F
#Adresy pozostale
NET_BRAMA=192.168.22.1
NET_MASKA=255.255.255.0
NET_DNS=192.168.22.1

Parametry dmuchawy:
DM_OFFSET = 0
DM_INWERSJA_PWM = 0
DM_MODULACJA = 0
#jesli 0 - brak detektora lub detektor przed bezpiecznikiem od peryferiow
DETEKTOR_TESTUJE_BEZPIECZNIK=0
Parametry modulacji (gdy DM_MODULACJA = 1):
DM_ROZRUCH_CZAS = 3
DM_ROZRUCH_MOC = 60
DM_PLUS_CZAS = 3
DM_PLUS_MOC = 20
DM_DOPALANIE_CZAS = 2
DM_DOPALANIE_MOC = 80
DM_TLO_MOC = 30

#przyklad mapowania czujnika dallas
# identyfikatory czujnikow dostepne pod adresem
# http://adres_ip_lucjana/dallas
#TPIEC=DALLAS[28FF265C64160379]

TPIEC=DALLAS[28FF060BA4150165]
TZEW=DALLAS[28FFFE67A7150327]
TPOWROT=DALLAS[28FFC815A41501C8]
TPODAJNIK=DALLAS[28FF09CBA3150149]
TCWU=DALLAS[28FF743AA7150314]
TPODLOGA=DALLAS[28FF6F16A4150122]

# mapowanie modulu termopary jeszcze nie podłączona termopara max6675
#TSPALINY = MAX31855[0]
TSPALINY = MAX6675[0]
# zmapowanie na pinie CS=48 i z wlaczonym filtrowaniem
TSPALINY = MAX6675[48.1]

#przyklad dodania offsetu -10.0 deg do T1
#OFFSET_T1=-100

zabezpieczenie przed cofaniem sie zaru
jesli TPODAJNIK wzrosnie powyzej T_MAX, uruchamia sie podajnik 
na czas wyrzutu paliwa z podajnika 5min. i uruchamia sie alarm
PODAJNIK_CZAS_WYRZUTU = 5
PODAJNIK_T_MAX = 60
jesli TPODAJNIK wzrosnie powyzej T_WYSOKA,
uruchamia sie cykle 30s (* max ilosc powtorek jesli ustawiono) podnoszenia zaru
PODAJNIK_CZAS_PODNOSZENIA = 30
PODAJNIK_T_WYSOKA_POWTORKI = 0
PODAJNIK_T_WYSOKA = 50

#czas w sekundach potrzebny do oproznienia pełnego zasobnika
PODAJNIK_CZAS_DO_OPROZNIENIA = 69000
#aktualny czas pozostaly do oproznienia zasobnika w sekundach
#UWAGA! po uruchomieniu zliczamy od tej wartosci
# (domyslnie 50% zasobnika) i tracimy poziom zasobnika!!
# poziom zasobnika ustawiac najlepiej przez WWW
#PODAJNIK_CZAS_PODAJNIKA = 34500
#procent ponizej ktorego ma wystapic alarm pustego zasobnika
#PODAJNIK_ALARM_MIN=15
#stała zalezna od opału w g./min.
PODAJNIK_PREDKOSC_PODAWANIA = 240

zabezpieczenie przed zbyt wysoka temperatura pieca
PIEC_T_MAX = 90

temperatura zalaczenia pomp
PIEC_T_MIN = 35

tryb pracy pieca RECZNY,AUTO
PIEC_TRYB = AUTO
jesli piec ma automatycznie startowac po uruchomieniu - ustawiamy AUTO
w trybie recznym mozna wlaczac (1) lub wylaczac urzadzenia (0)
#OUT_PODAJNIK=1
#OUT_POMPACO=0
#OUT_POMPACWU=0
#OUT_POMPACWU2=0
#OUT_POMPACYRK=0
#moc wentylatora 50%:
#OUT_WENTYLATOR=50

algorytm pracy palnika - ECOAL,RR,RRM,RRM2,ZASYPOWY,WYLACZONY,...
PIEC_ALGORYTM = RRM2
PIEC_ZADANA = 75
uwaga histereza pieca * 0.1deg
PIEC_HISTEREZA = 0

#parametry algorytmu RR, czas podtrzymania w minutach, reszta w sekundach:
#RR_PODTRZYMANIE_CZAS = 10
#RR_PODTRZYMANIE_PODANIE = 10
#RR_PODTRZYMANIE_POSTOJ = 30
#RR_PODTRZYMANIE_MOC = 40

#RR_PRACA_PODANIE = 2
#RR_PRACA_POSTOJ = 5
#RR_PRACA_MOC = 23

#rrm 
# 1 - wlaczone tlo, 2 - dopalanie
RRM_TLO_WLACZONE=1
RRM_TLO_MOC=20
RRM_TLO_CZAS_PODANIE=10
# w min.
RRM_TLO_CZAS=15
#jesli tlo podbije TPIEC 5deg ponad zadana przechodzimy w podtrzymanie
RRM_MAX_PRZEBICIE_TLO=5
# dopalanie 150s 30%
RRM_DOPALANIE_MOC=30
RRM_DOPALANIE_CZAS=150

#PONIŻEJ DODANY MODÓŁ WIELOMOCOWY  RRM2

# 1 - tryb wielomocowy dla RRM2
RRM_WIELE_MOCY=1
# moce od 0 (najmniejsza) do 5 (najwieksza)
# moc 0 domyslna
RRW_PODANIE0=4
RRW_POSTOJ0=18
RRW_NADMUCH0=60
# moc 1
RRW_PODANIE1=3
RRW_POSTOJ1= 15
RRW_NADMUCH1=70
RRW_HISTEREZA1=20
# moc 2
RRW_PODANIE2=3
RRW_POSTOJ2= 12
RRW_NADMUCH2=80
RRW_HISTEREZA2=40
# moc 3
RRW_PODANIE3=3
RRW_POSTOJ3= 10
RRW_NADMUCH3=90
RRW_HISTEREZA3=60
# moc 4
RRW_PODANIE4=4
RRW_POSTOJ4= 9
RRW_NADMUCH4=100
RRW_HISTEREZA4=80

tryb pracy pompy co (ECOAL,BRULI,ZIMA,LATO)
CO_TRYB = ZIMA
algorytm pracy pomy CO (CIAGLY,CYKLICZNY)
CO_ALGORYTM = CIAGLY

sargus123, 2017-10-30:
miałem przydzielone na sztywno w routerze. po mac. :)  ale i tu wpisać nie zaszkodzi.

uzi18, 2017-10-30:
Potwierdz mi tylko ze opcja dziala jak oczekiwales

Amozeja, 2017-10-31:
Na razie nie potwierdzę, bo u mnie Lucjan na razie jeszcze w kawałkach, a po prostu coś takiego mi się nasunęło w czasie obsługi fabrycznego sterownika.

Amozeja, 2017-11-01:
Lucjana znam na razie tylko z teorii umieszczonej tu i na elektrodzie i esterowniku, więc mogę się mylić, A chodzi mi o bezwarunkowe zachowanie sterownika przy każdym włączeniu, niezależnie, czy to jest lato, czy zima. Tak, aby po pierwsze: zapewnić większe bezpieczeństwo w przypadku, gdy braknie prądu, ale ciąg kominowy powoduje jeszcze palenie się paliwa i ewentualny wzrost temperatury kotła (co prawda do momentu wypalenia się paliwa, ale trzeba byłoby przewidzieć zaistnienie sytuacji nawet praktycznie niemożliwej). Po drugie: start z zamkniętym zaworem czterodrogowym, z tych samych względów nie jest chyba najlepszym rozwiązaniem.  
P.S.
Mój Lucjan na razie jest "w kawałkach", brakuje mi trochę drobnicy, sterowania dmuchawą, pomysłu na obudowę i czegoś tam jeszcze. Ostatnio, po pewnych kłopotach, udało mi się zaprogramować ESP. Najpierw próbowałem z ESP-01, ale bez skutku. Udało mi się z NodeMCU.
 No i problem mam z ethernetshieldem Arduino - mój ruter (Livebox niestety) nie widzi urządzenia. Jeśli stworzę inną sieć z wykorzystaniem innego rutera - jest widoczny. Nie jest to bynajmniej problem z ustawieniami konfigu w tym zakresie. Nie pomaga nawet zmiana maca karty.

uzi18, 2017-11-02:
Net_ip w trybie auto czy z konkretnym adresem?
W trybie auto router powinien go widziec.
Chociaz livebox bywa kaprysny.
Do podlaczenia nowego urzadzenia chyba jakis przyciak trzeba wcisnac.

Polecam do sterowania pompami i dmuchawa modul triakow z podmienionym
MOC-em plus detektor zera sieci.

uzi18, 2018-10-17:
update?