uzi18/sterownik #84: Dodanie obsługi podajnika tłokowego

tjtm, 2017-10-31:
Dla obsługi podajnika tłokowego potrzebnych będzie kilka zmian. Na początek:
1. Obsługa czujnika położenia tłoka. Czujnik ten pokazuje czy tłok znajduje sie w pozycji startowej i można rozpocząć cykl podawania. Po uruchomieniu podawania czujnik ten informuje o powrocie podajnika do pozycji startowej, a co za tym idzie o zakończeniu podawania.
Czyli tak:
Sprawdzić czy tłok w pozycji startowej, jeżeli tak, to start podawania, sprawdzanie czy podaj nik wrócił do pozycji startowej, jeżeli tak, to przerwać cykl podawania.
2. Potrzebny parametr określający maksymalną długość cyklu podawania. Jeżeli czas pomiędzy startem podawania, a końcem przekroczy ten parametr, to ALARM - problem z podajnikiem.
Przykładowo normalny cykl tłoka trwa ok 40s, parametr ustawiony na 60s, jeżeli przekroczony, to coś się stało z podajnikiem.
3. Parametr określający ilość cykli podajnika po przekroczeniu temperatury podajnika/zasobnika. Zepchnięcie żaru do popielnika.
Tyle na początek. Myślę, że w testach coś się jeszcze pojawi.

uzi18, 2017-10-31:
v0.1.0.130.3 - wstepne wsparcie dla podajnika tlokowego
    - dodanie parametrow dla podajnika tlokowego
    - poprawka zakresu czasu dla naweglania

uzi18, 2017-10-31:
Ad.1 skonczone