uzi18/sterownik #85: pomiar paliwa w zasobniku.

sargus123, 2017-11-12:
zauważyłem na jednej stronie odczyt ilości paliwa za pomocą sensora odległości. 

może coś takiego zrobić. wtedy będzie zawsze pewne jaki jest faktyczny poziom paliwa w zasobniku. 

http://projektpimalina.blogspot.com/2016/05/czujnik-poziomu-wegla-na-nadajniku.html 

w pliku config można wstawić  pozycję gdzie się podaje wysokość zasobnika od czujnika aby poprawnie odczytywał poziom w każdym zasobniku.

uzi18, 2017-11-13:
Byl juz taki pomysl na poczatku. Nikt nie sprawdzil jednak dotad czy zapylenie/zadymienie i ogolne warunki w zasobniku nie powoduja niszczenia tego czujnika.

sargus123, 2017-11-13:
sprawdzę czujnik ultradźwiękowy w obudowie szczelnej. samochodowy one są po 8 zł.  sprawdzę ich zakresy działania. one się nie kurzą itp.

uzi18, 2017-11-13:
Nie wiem jak z podlaczeniem tego samochodowego do mikrokontrolera

sargus123, 2017-11-15:
To co miałem w planie nie wyszło . za niski sygnał daje nadajnik i głupoty wychodzą w odczycie. 
ale znalazłem hermetyczny już gotowy moduł.  ok. 55zł. - JSN-SR04T 
Czujnik jest na kablu. pomiar od 20 cm. w dal. ale patrząc na głębokość zasobnika to jest do przyjęcia.

uzi18, 2017-11-15:
To mi sie podoba,  konkretny modul i sonda ktora na pierwszy rzut oka moze wytrzymac warunki.
Jest jeszcze tylko jedna rzecz w zasobniku panuje nadcisnienie,  ciekawe czy to nie uszkodzi czujnika albo nie bedzie wplywac na pomiary.
Tak czy inaczej wymagane sa testy =)

uzi18, 2017-11-15:
@bzykbzyk, @ravikjab,  @Stan17,  @mark3k3 przyda się?

sargus123, 2017-11-15:
czujnik ten nie jest wrażliwy na ciśnienie tym bardziej kiedy będzie zamontowany wewnątrz.  te czujniki są montowane w uniwersalnych czujnikach samochodowych.

Stan17, 2017-11-16:
To ja kiedyś poruszałem ten temat i dokładnie na podstawie tego samego znaleziska na projektpimalinie.
Ale mieliśmy wtedy inne problemy na głowie, i projekt został odłożony na przyszłość.
Myślę że warto do niego wrócić, bo nie wiem jak Wy ale ja nie raz zapomniałem zresetować licznika :)

uzi18, 2017-11-16:
W sumie moze ten pomiar pozwolilby na automatyczne "uzupelnienie" paliwa

sargus123, 2017-11-16:
Jeszcze jedno co może układ spełniać. 
Po założeniu w klapie to można w ustawieniach dołożyć kod że jeśli odległość jest powyżej głębokości zbiornika to ma wyłączyć wentylator i podajnik a włączyć po  np. 30 sekundach kiedy odczyt będzie prawidłowy. (ponieważ w czasie ładowania do zbiornika będzie "widział nas" jako odległość bliższa niż głębokość zbiornika.) takie rozwiązanie zautomatyzowało proces ładowania  / czyszczenia pieca.

uzi18, 2017-11-16:
@Stan17  czyli co zamawiamy u Ali-ego?

uzi18, 2017-11-16:
@bzykbzyk zglaszal jeszcze cos takiego kiedys:
#45

uzi18, 2017-11-16:
klawisz wywołujący zwloke w dalszej pracy na czas wyczyszczenia pieca #80

uzi18, 2017-11-16:
Duplicate of #45

bzykbzyk, 2017-11-16:
JSN-SR04T na Ali za 8 baksów. Zamawiam. Jak nie zadziała w sterowniku zabuduję w samochodzie...

uzi18, 2017-11-16:
Tez zamowilem na ali.

Tak swoja droga tutaj mamy nasz online chat: https://gitter.im/sterownik/Lobby
@Stan17 @mark3k3 @ravikjab @bzykbzyk @micro-jr @ebyna @Kosiarz78

Stan17, 2017-11-17:
No to zamawiamy , ten JSN-SR04T ?

uzi18, 2017-11-17:
Tak one są zgodne z tymi użytymi w linku z 1 postu

uzi18, 2017-12-29:
Doszly czujniki do mnie, najpierw dodam pomiar kazdy sobie przetestuje czy dziala mapujac czujnik na wykres

uzi18, 2018-01-03:
OK Panowie ciekawostka modul JSN-SR04T-2.0 sam potrafi mierzyc odleglos i wysylac ja po RS232/TTL

uzi18, 2018-01-03:
@sargus123 @Stan17 @bzykbzyk teraz pytanie czy zaprogramowac stary tryb z pomiarem echa czy nowy - odczyt pomiaru z czujnika (wymaga przylutowania rezystora) czy oba ?

sargus123, 2018-01-04:
tak myślę że dobrze by było aby układ sam podawał dane a z configu będzie łatwiej podać poziomy:
- pełny,  
- pusty , 
-klapa otwarta. 
po co miałby nasz lucjan jeszcze obliczać. (może to zmniejszy objętość kodu.) :)
ale to nie ja będę pisał kod :) ważne by przyszły nowy użytkownik nie miał problemu ze złożeniem. 
problemy gdzieniegdzie widziałem z wersją 2.0 
Przykładowy kod działa tutaj zarówno dla wersji V1.0, jak i wersji 2.0
https://www.dfrobot.com/wiki/index.php/Weather-proof_Ultrasonic_Sensor_SKU_:_SEN0207

uzi18, 2018-01-04:
@bzykbzyk Ty tez masz ten w wersji 2.0 ?

Stan17, 2018-01-04:
Ja mam wersję 2.0
Ale to nam zablokuje RSa, ja mam tam podłączonego ecoala i chciałbym tak zostawić puki co.
Tak więc ja głosuję implementacją obu metod.

uzi18, 2018-01-04:
mamy jeszcze 1 wolny RS

uzi18, 2018-01-04:
uzywasz esp-link?

bzykbzyk, 2018-01-04:
@uzi18 jeszcze nie mam czujnika, zamówiłem https://www.aliexpress.com/item/K02-JSN-SR04T-integrated-ultrasonic-ranging-module-reversing-radar-40KHZ-waterproof-ultrasonic-sensor-module/32732427963.html?spm=a2g0s.9042311.0.0.OUlc6D, na foci płytki jest 2.0. Kitajce coś się ociągają z moimi dostawami, na razie mam komunikat, że przygotowane do wysyłki... Chyba tam pojadę i im nawrzucam....

sargus123, 2018-01-04:
ja nie używam. esp link na lucjanie.

Stan17, 2018-01-05:
Ja też jeszcze nie.

uzi18, 2018-01-06:
@Stan17 czyli mozemy uzyc TXD2/RXD2 ??

uzi18, 2018-01-06:
```
SR04T: 127 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 123 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 119 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 115 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 111 FF 2 27 28,STX,CRC,227 551mm.
SR04T: 107 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 103 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 99 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 95 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 95 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 91 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 87 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 83 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 79 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 75 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 71 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 67 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 63 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 63 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 59 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 55 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 51 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 47 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 43 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 39 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 35 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 31 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 27 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 23 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 19 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 15 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 11 FF 2 16 FF,STX,CRC zly
SR04T: 8  S, S, S,FF 2 16 17,STX,CRC,216 534mm.
SR04T: 8 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 8 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 8 FF 2 16 17,STX,CRC,216 534mm.

SR04T: 127 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 123 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 119 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 115 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 111 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 111 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 107 FF 2 27 28,STX,CRC,227 551mm.
SR04T: 103 FF 2 27 28,STX,CRC,227 551mm.
SR04T: 99 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 95 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 91 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 87 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 83 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 79 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 75 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 75 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 71 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 67 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 63 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 59 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 55 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 51 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 47 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 43 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 43 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 39 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 35 FF 2 16 17,STX,CRC,216 534mm.
SR04T: 31 FF 2 16 17,STX,CRC,216 534mm.

Cos juz dziala :)
```

uzi18, 2018-01-06:
v0.1.0.130.11
        - dodanie obslugi pomiaru z czujnika SR04T-2.0
          podlaczamy TX czujnika pod RX2 na plytce Mega
          w pole R27 modulu wstawiamy rezystor 47k
        - dodanie mozliwosci zmapowania poziomu zasobnika na wykresie
          np. T4=SR04T[]
        - LCD: dodanie pomiaru odleglosci na ekranie poziomu zasobnika

uzi18, 2018-01-06:
testujcie sobie i opiszcie wnioski tutaj :)

sargus123, 2018-01-08:
[image]
u mnie bez zarzutu działa.

sargus123, 2018-01-08:
z tego co zauważyłem tonie pokaże poziomu poniżej 18.5 centymetra.gdy jest więcej załadowane. ale tak naprawdę to w niczym nie przeszkadza.  2/3 zbiornika mi opróżniło i wyświetliło poprawnie.

sargus123, 2018-01-09:
u mnie 60 to pusty a 65 ślimak goły.

uzi18, 2018-01-09:
Super czekamy teraz na informacje od  reszty testerów :)

sargus123, 2018-01-10:
montaż czujnika 
puszka uniwersalna. 
W pokrywie wywiercone 3 otwory 2 na śrubki 3mm i 1 ma kabel 8mm . nie uszczelniałem ponieważ od spodu puszka równo przylega do blachy i czujnik w obudowie. 
jeśli będzie trzeba to się uszczelni od góry pod pokrywką.

[image] 

Warto zwrócić uwagę na to  że czujnik nie patrzy w pełni na wprost. Sonda jest lekko pod skosem. 
Trzeba tak umieścić aby poprawnie patrzył w dno zbiornika . 

[image]
 zostało jeszcze poprowadzenie kabla w osłonie.

uzi18, 2018-01-10:
Zabezpiecz przewod aby sie nie łamał prxy otwieraniu klapy

Stan17, 2018-01-10:
Czym z wierzchu przykryłeś ten kabel ?

sargus123, 2018-01-10:
Uniwersalna puszka ze sklepu elektronicznego ma pokrywkę. Nią przykryłem
usuwając tylko w miejscu kabla krawędź aby kabla nie przygniotlo.

sargus123, 2018-01-10:
Potem kabel jeszcze zabezpieczenie w sporo aby luzem nie latał tam gdzie
nie powinien.

10.01.2018 1:00 PM "Artur Kuchnowski" <sargusix@gmail.com> napisał(a):

sargus123, 2018-01-17:
Drobna uwaga warto pod szpilki układu dołożyć kondensator układ jest wrażliwy na zasilanie. 
Miałem z tym układem problem że układ nie chciał wskazywać większej  odległości niż 33 centymetry. 
Po podłączeniu pod zasilanie problemy ustąpiły.

Stan17, 2018-01-19:
Masz na myśli jakiś elektrolit na +- zasilania ?

uzi18, 2018-01-19:
Tak

uzi18, 2018-01-19:
Tak bezposrednio na module,  miedzy gnd i vcc.
@sargus123 a jaki kondzior dales 1000uF?

sargus123, 2018-01-19:
ja wrzuciłem 829 uf pod + i -

sargus123, 2018-01-19:
820

uzi18, 2018-02-02:
W aktualnej podwersji... 21  dodalem kontrole stanu minimalnego