uzi18/sterownik #87: Menu sterownika lcd

sargus123, 2017-11-25:
Mam propozycję układu menu  w pdf. 
mam nadzieję że łatwo będzie się rozczytać. i co do pamięci to można dołożyć pamięć na szynę I2C i też tam dane upchać ceny są ok. 9 zł 

[menu sterownika.pdf](https://github.com/uzi18/sterownik/files/1503469/menu.sterownika.pdf)

To tylko propozycja :) ale pozwoli na zarządzanie z menu prawie wszystkim

uzi18, 2017-11-25:
Ok zerkne na to

uzi18, 2017-11-27:
@Stan17 @ravikjab @mark3k3 @wwpiotr @mipurel co o tym sadzicie?

26.11.2017 12:06 AM "Bartłomiej Zimoń" <cactus.list@gmail.com> napisał(a):

Stan17, 2017-11-28:
Sam pomysł schematu wygląda ok.
Trzeba by rozrysować w całości żeby wyłapać jakieś niekonsekwencje czy nieścisłości hierarchiczne ale tak na oko wygląda że jest to dobry pomysł.

uzi18, 2017-11-28:
Pamiętam tylko ze nie chcialem wrzucac tam wszystkiego jest ogrom opcji

sargus123, 2017-11-28:
Ale z tego co kojarzę To można menu konfiguracyjne powiązać z plikiem
tekstowym. aby pobierał tabelę i dane z niego lub pamięci. I przerzucić
część danych co można zapisywać i tworzyć. Coś jak baza danych.

Trochę pisania jest ale za To pełna wygoda. Jak w oryginalnym sterowniku.

Stan17, 2017-11-28:
hehe, Bartek ostatecznie i tak to Ty decydujesz, bo w końcu będzie to Twoja praca w to włożona.
Ja tylko odpowiedziałem że wydaje mi się to sensowne.

uzi18, 2017-11-28:
Ale to jest oryginalny sterownik =)
Np.  Parametry w pliku to dosc oryginalne rozwiazanie.

sargus123, 2018-01-05:
Np. Parametry w pliku to dosc oryginalne rozwiazanie. - miałem na myśli możliwość konfigurowania tych parametrów z menu z zapisem do pamięci - pliku