uzi18/sterownik #90: Czujnik CO2

Mirek3012, 2018-01-03:
Oby się nie przydał ale...

uzi18, 2018-01-03:
https://abc-rc.pl/Czujnik-MQ-7

uzi18, 2018-01-04:
Duzo lepsza alternatywa - Satel DG-1 CO koszt ok. 200PLN

Stan17, 2018-01-04:
No nie wiem, mówimy o bezpieczeństwie a w takim razie moim zdaniem czujnik CO powinien być niezależnym osobnym urządzeniem.

Mirek3012, 2018-01-05:
Przepraszam zakladjac temat nie dokładnie opisałem, Raczej czujnik CO Lucjana miał posłużyć jako dodatkowe urządzenie. Dlatego mam 2 czujniki , jeden w kotłowni w piwnicy drugi na parterze, niestety ten z kotłowni nie przebija się z dźwiękiem na parter . Tylko jest pytanie w jakim stopniu można ufać market owym czujniką , I jak sprawdzić poprawność działania po kilku latach. Niby jest przycisk test ale z tego testu dowiemy sie ze buzzer pracuje prawidlowo