uzi18/sterownik #91: Rozdzielenie config na kilka części.

sargus123, 2018-01-05:
możne trzeba rozdzielić config na 2 części albo i 3 czy 4
siec, sprzęt, parametry , typ_pracy

moja propozycja : 

- sieć  /ustawienia sieciowe , ustawienia dostępu do lucjana Hasła itp.

-sprzęt / ustawienie czujników dallas, termopary, czujnika zera, czujnik poziomu paliwa

- parametry / ustawienie podajnika, temperatury minimalnej, maksymalnej pieca , temperatura załączania pomp, czas wyrzutu paliwa, ustawienie poziomu paliwa w zasobniku, lub ustawienie czasu na opróżnienie zasobnika. działanie zaworu 4 drogowego, podłogówki, 

- typ_pracy  tu każdu inny plik w zależności od wybranej metody sterowania  pliki o nazwach : ECOAL ,RR, RRM, RRM2, ZASYPOWY, WYŁĄCZONY i w nich ustawienia od tych typów. albo jeden wspólny plik do każdego typu.

ważne będzie opisanie danych jakie i w jakich plikach się znajdują..

Mirek3012, 2018-01-05:
Jestem za config bylby czytelniejszy

uzi18, 2018-01-06:
Duplicate of #23 

Tu bardziej rozszerzona wersja.

uzi18, 2018-01-06:
Duplicate of #23