uzi18/sterownik #95: zabezpieczenie przed awarią czujnika temperatury.

sargus123, 2018-01-15:
witam.
 Dopisać trzeba było zabezpieczenie na wypadek awarii termometru od pieca. 

wypiął się czujnik a lucjan wyłączył pompę obiegową i kotłownie zalało. Ponieważ piec dalej się palił.

uzi18, 2018-01-17:
Zacznijmy od tego ze czujnik przegrzania kotla by Cie przed tym zabezpieczyl.
Zamontuj go bezwarunkowo.

Ponizej wrzucam wsad do przetestowania.
Trzeba sprawdzic czy mierzy poprawnie czujniki i co sie dzieje gdy je odlaczysz.

uzi18, 2018-01-17:
[Lucjan.hex.zip](https://github.com/uzi18/sterownik/files/1637377/Lucjan.hex.zip)

uzi18, 2018-01-17:
v0.1.0.130.15
        - max czas dla podtrzymania to 240min.
        - zapamietujemy stan dla alarmu tmin po resecie
        - Dallas: alarm gdy brak czujnika lub bledy komunikacji
        - nowy alarm uszkodzonego czujnika
        - Dallas: wykrywamy sytuacje braku czujnika/czujnikow

    - Domoticz: poprawki dzialania w trakcie aktualizacji procesora

closed 2018-01-17 by uzi18