uzi18/sterownik #103: Poprawić obsługę alarmów czujników

mecenas8, 2018-02-08:
Po ostatnich zmianach w sofcie następuje przejście w tryb  ręczny, co jest problemem ponieważ zatrzymuje dmuchawę i podajnik. Problem zauważyłem ponieważ okazało się, że sporadycznie zdarza się błąd odczytu czujnika TZEW. Najlepiej byłoby, żeby przy pojedynczych sporadycznych błędach w odczycie czujnika zgłosić alarm ale pozwolić na kontynuację procesu spalania.

uzi18, 2018-02-08:
Wklej zawartosc podstrony /dallas

mecenas8, 2018-02-08:
DALLAS 00 ID:28CCE6B501000001 p:1 o:0 m:1 e:0 t:409
DALLAS 01 ID:285AB6B60100001F p:1 o:0 m:1 e:0 t:465
DALLAS 02 ID:28A3E1A601000007 p:1 o:0 m:1 e:0 t:593
DALLAS 03 ID:28AF0DA70100000E p:1 o:0 m:1 e:0 t:8
DALLAS 04 ID:0000000000000000 p:0 o:0 m:0 e:0 t:0
DALLAS 05 ID:0000000000000000 p:0 o:0 m:0 e:0 t:0
DALLAS 06 ID:0000000000000000 p:0 o:0 m:0 e:0 t:0
DALLAS 07 ID:0000000000000000 p:0 o:0 m:0 e:0 t:0
DALLAS 08 ID:0000000000000000 p:0 o:0 m:0 e:0 t:0
DALLAS 09 ID:0000000000000000 p:0 o:0 m:0 e:0 t:0
DALLAS 10 ID:0000000000000000 p:0 o:0 m:0 e:0 t:0
DALLAS 11 ID:0000000000000000 p:0 o:0 m:0 e:0 t:0
DALLAS 12 ID:0000000000000000 p:0 o:0 m:0 e:0 t:0
DALLAS 13 ID:0000000000000000 p:0 o:0 m:0 e:0 t:0
DALLAS 14 ID:0000000000000000 p:0 o:0 m:0 e:0 t:0
DALLAS 15 ID:0000000000000000 p:0 o:0 m:0 e:0 t:0

Niestety wkleiłem z wersji bez alarmów - 0.1.0.130.11, ponieważ na takiej akurat pracuję, gdyż nie stwarza problemów.
Na nowszej wersji wyglądało podobnie, a w czasie alarmu przy danym dallasie pojawiało się e:1.
Problem z moim czujnikiem może polegać na tym, że jest zasilany dwuprzewodowo (Parasite power), jest kilka metrów przewodu,
 na razie nie miałem czasu i chęci by wymienić go na 3-przewodowy. 
Jednak dla bezpieczeństwa takie sporadyczne błędy raczej nie powinny zatrzymywać na stałe pracy kotła. 
Kocioł pracował może 20-40 minut zanim pojawiał się chwilowy błąd, który wywoływał alarm i blokował spalanie.

uzi18, 2018-02-08:
Jak mozesz to zrob na nowej wersji i wklej mi dodatkowo fragment konfigu z przypisaniami tych czujnikow

uzi18, 2018-02-08:
5 błędów powoduje alarm,  kasują się w momencie poprawnej komunikacji.
Moze u Ciebie tych bledow jest wiecej.

sargus123, 2018-02-08:
Nie wiem czy to pomorze ale miałem problem z czujnikami jak miałem za cienki przewód do nich poprowadzone. 
Powodowało u mnie sporadyczne problemy z odczytem czujników nieraz termometr zawieszał się na domyślnej 85 stopni.

Miałem2 żyłową instalację w postaci gwiazdy 50m. ale ważny był przekrój przewodu. 
gdzie nie gdzie piszą o zmianie opornika. no i napięcie stabilne.

uzi18, 2018-02-08:
Pojemnosc przewodu wplywa na jakosc komunikacji, na dluzsze odcinki polecane zwykle byly koncentryki/skretka UTP a nawet przewod 2,5mm2 ;)

uzi18, 2018-12-16:
wg dokumentacji najlepiej uzywac skretki oraz instalacji 3 przewodowej