uzi18/sterownik #107: Pellet na Lucjanie ?

kmarcin84, 2019-10-31:
Czesc.
Gratuluje projektu.
Mam nadzieje ze koledzy beda mi w stanie pomodz w rozwiazaniu "problemu".
Jestem w trakcie skladania "Lucjana", ale od poczatku. 
Postanowilem przerobic swojego smieciucha na pellet , ew groszek. 
W tym celu powstal juz odpowiedni podajnik z palnikiem, teraz kwestia sterownika. 

Pytanie:

1. Czy jest mozliwosc konfiguracji w ktoryms z algorytmow spalania trybu : "rozpalanie" , gdzie zalaczy mi odpowiednie wyjscie przekaznikowe na okreslony czas? Chodzi tutaj o zalaczenie grzalki ceramicznej przy automatycznym rozpalaniu pelletu.

Np:
- Podajnik powiedzmy 3 sekundy
- zalacz grzalke ceramiczna na xx 
- zalacz dmuchawe na powiedzmy 40%
- odczyt temperatury z termopary (zamontowana pod palnikiem na pellet)
- wylacz jezeli temp > xx st C

2. Czy mozna odpowiednie wyjscie przekaznikowe/tyrystorowe konfigurowac z pliku TXT z karty SD, lub czy istnieje jakis spis akceptowanych przez sterownik komend ?

3. Czy algorytmy dmuchawy AC wymagaja ukladu na 2008 czy wystarczy tyrystorowy regulator (taki posiadam i dziala swietnie od min - max na posiadanej dmuchawie) - postaralbym sie ewentualnie dolaczyc sterowanie PWM z arduino do tego.

Bede wdzieczny za pomoc w temacie.

Pozdrawiam.

uzi18, 2019-10-31:
musze to przemyslec ale rozpalanie moge sprobowac Ci dodac jak znajde chwile z czym ostatnio krucho, na kiedy to potrzebujesz?

kmarcin84, 2019-10-31:
Jestem na etapie zabudowy elektroniki w obudowe rozdzielnicy i od odpalenia palnika dzieli mnie juz tylko podlaczenie sterownika wlasciwie. 
Nie ukrywam że zależałoby mi na tej opcji jeżeli miałbyś chwilę.. Było by super.

Dmuchawa to AC, jakis standard z kondensatorem w srodku.

Tymczasowo mogę "rozpalarkę" wyrzucic na panel jakimś switchem choć kiedyś chciałbym zrobić to w automacie, miec mozliwosc sterownia np przez www.
Jezeli uda mi sie to odpalic to wrzucę na elektrodę kilka fotek i sprawozdanie. 

Działam na ATMega+Wifi (Z esp8266 na jednej plycie) + eth shield, RTC shield jakis mam, ale musze eeprom dolozyc (po co on wogole?), do tego kilka dallasow : piec+cwu+tpodajnik prawdopodobnie + twew - tutaj jakis termostat jeszcze doloze do pokoju kiedys) no i 2 termopary - jedna w palniku druga na czopuch.  Zadnych zaworow 4d, jeden obieg.

uzi18, 2019-10-31:
na razie daj przycisk, ogolnie ze strony www da sie uruchomic piec, wystarczy ja troszke zmodyfikować - dodac przyciski

WojtekGD, 2019-11-04:
Cyt: "Zamontowalem jeszcze fotorezystor w drzwiczkach opodal palnika, ma wykrywac obecnosc plomienia."
To może się nie sprawdzić w piecu na pellet czujnik może dosyć szybko się brudzić - lepiej zrobić algorytm oparty o temperaturę paleniska - termopara + temp. spalin druga termopara.

uzi18, 2019-11-04:
zgadzam sie palenisko to nie jest miejsce na czujniki foto

kmarcin84, 2019-11-04:
Spokojnie koledzy.
Czujnik zamontowany jest nie w palenisku tylko w drzwiczkach.
Od drzwiczek odchodzi 5 cm rurki.
W rurce silikon temp》300°C > fotorezystor.
Dodatkowo odmuchiwany jest powietrzem a w drzwiczkach szklane oczko :)
Nie spali się..

Mozna tutaj foto dodac? Czy moze na jakims forum? [#esterownik.pl/elektroda.pl].
Jutro wezmę się za zabudowanie w rozdzielnicy elektroniki / może triaki na uniw. pcb polultuję - jak czas pozwoli...

uzi18, 2019-11-04:
wrzuc tu i tam :)

WojtekGD, 2019-11-04:
Dyskutujemy raczej tutaj: https://gitter.im/sterownik/Lobby?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge&utm_content=badge co do czujnika nie tyle chodzi o jego zniszczenie co o zabrudzenie. Czujniki optyczne są stosowane w palnikach olejowych i gazowych ale tam przy prawidłowym wyregulowaniu palnika zanieczyści się po bardzo długim używaniu i spaliny trudno nazwać dymem. Dodatkowo umieszczony z reguły jest za deflektorem a tam trudno go zabrudzić jeśli palnik pracuje prawidłowo.

uzi18, 2020-02-23:
ok racja poprawilem ;)