uzi18/sterownik #109: Nie problem ale pomysl na ewolucje

suenio, 2020-07-02:
W zwiazku z rozwojem ESP32 pomyslalem sobie ze fajnie byloby wykorzystac np R1 32 zamiast Arduino Mega jako baze. Jest to plytka ala UNO na bazie ESP32 ktory daje polaczenie wifi/BT + wystarczajaca ilosc GPIO (Digital and Analog). Chetnie bym sie w to zaangazowal gdyz wlasnie planuje budowe sterownika w oparciu o arduino i "Lucjana" dodajac do tego integracje z BLYNKiem.

uzi18, 2021-01-03:
@suenio  Przeciez bramke do BLYNK mozesz na bazie esp32 stworzyc w kazdej chwili, komunikowac sie z Mega przez rs232 i przesylac do i z blynk

daj link do plytki

suenio, 2021-01-04:
Tak oczywiscie mozna dodawac plytek coraz wiecej tylko pomysl byl ze skoro jest juz esp32 zamiast esp8266 to ma na tyle wyjsc/wejsc ze mozna zrobic to wszystko na jednej plytce czyli same I/O + bezprzewodowe polaczenie z internetem chodzi np o plytke 
https://pl.aliexpress.com/item/1005001621901111.html
Sa tez wersje mini czy inne wersje esp32 ale to jest gabarytowo standardowe "UNO" wiec ogolnie latwiej :)

uzi18, 2021-01-04:
Juz o tym kiedys byla dyskusja, musielibysmy zrobic przynajmniej jeden moduł do sterowania dmuchawa w formie minimodulu. Wyjscia cyfrowe musialyby isc przez jakies ekspandery bo ich jest tu zdecydowanie za malo i wtedy to zaczyna miec sens.
Musialbym popatrzec bo moze cos sie zmieniło... poza tym jesli brac cos pod uwage to tylko ta bagatsza wersja esp32 a ciekawe jaka jest tutaj.

uzi18, 2021-01-04:
Chodzi o to ze musialby byc esp32 w wersji 2 rdzeniowej, bo jest kupa zabawy z tym aby wifi nie robilo balaganu i sie nic nie wieszali, przerabialismy to wszystko w projekcie ESPEasy.

suenio, 2021-01-04:
Ta plytka do ktorej podalem link jest w wersji 2 rdzeniowej. Posiadam na stanie i uzywam w kosiarce autonomicznej bo racja ze wifi samo z siebie a Blynk szczegolnie trzebaby osadzic na osobnym rdzeniu w stosunku do funkcji sterujacej piecem. Co do sterowania dmuchawa to nie do konca rozumiem bo PWM powinien wystarczyc ale moze sie myle. A co do wyjsc cyfrowych to zakladajac 5 pomp obiegowych i 10 czujnikow temperaturowych to jest 15 a wlasnie ten modul ma ich ponad 20scia w odroznieniu do starych ESP8266 ktore mialy krucho z wyjsciam IO. Ame moze cos pomijam co lucjan potrzbuje jesli chodzi o I/O

uzi18, 2022-02-24:
@suenio fakt te esp32 sa dosc ciekawe i troche juz stabilniej dzialaja