uzi18/sterownik #110: Wygaszacz wyświetlacza

WojtekGD, 2020-12-28:
Chyba fajna opcja oszczędzania wyświetlacza - wygaszacz.
Dodanie jednego parametru np: 
wygaszacz = 
 tu wiadomo
0 - nie działa i wyświetlacz swieci jak świecił
1-.. wartość w sekundach lub minutach i wyświetlacz gaśnie

uzi18, 2021-01-03:
```
#po ilu sekundach ma sie wygasic wyswietlacz
# zakres 0-200, gdzie 0 - nie wygaszamy
#LCD_WYGAS = 0

```

test w zalaczeniu

[Lucjan.hex_33s.zip](https://github.com/uzi18/sterownik/files/5762042/Lucjan.hex_33s.zip)

WojtekGD, 2021-01-15:
Działa wygaszacz ale przestały działać termostaty mój config:

AUTOLATO_TEMP = 12 #pompa CO
AUTOLATO_TWEW = 21 #pompa CO
AUTOLATO_HISTEREZA = 10
AUTOLATO_TRYB = 0
UT1_ZADANA = 24   #SALON 24/20
UT2_ZADANA = 24   #LAZIENKA 25/20
UT3_ZADANA = 19   #WIATROLAP 19/15
TWEW_ZADANA = 22 #obnizenie temp CO
TWEW_HISTEREZA = 10
TWEW_CO_OBNIZONA = 70

PIEC_T_MAX = 90
PIEC_T_MIN = 40
PIEC_ALARM_T_MIN = 40
PIEC_TRYB = AUTO

PIEC_ALGORYTM = RRM2 # brykiet=RRM2 olej=RR
PIEC_ZADANA = 55
PIEC_HISTEREZA = 10

RR_PODTRZYMANIE_CZAS = 30
RR_PODTRZYMANIE_PODANIE = 8 #5
RR_PODTRZYMANIE_POSTOJ = 10
RR_PODTRZYMANIE_MOC = 20
RR_PODTRZYMANIE_CZAS_LATO = 600
RR_PRACA_PODANIE = 1
RR_PRACA_POSTOJ = 20
RR_PRACA_MOC = 100

RRM_TLO_WLACZONE = 0
RRM_WIELE_MOCY = 1

RRW_HISTEREZA0 = 10
RRW_PODANIE0 = 2
RRW_POSTOJ0 = 50
RRW_NADMUCH0 = 25
RRW_HISTEREZA1 = 20
RRW_PODANIE1 = 2
RRW_POSTOJ1 = 45
RRW_NADMUCH1 = 30
RRW_HISTEREZA2 = 40
RRW_PODANIE2 = 2
RRW_POSTOJ2 = 40
RRW_NADMUCH2 = 35
RRW_HISTEREZA3 = 70
RRW_PODANIE3 = 2
RRW_POSTOJ3 = 35
RRW_NADMUCH3 = 40
RRW_HISTEREZA4 = 90
RRW_PODANIE4 = 2
RRW_POSTOJ4 = 30
RRW_NADMUCH4 = 45

UT_WSPOLNY_PIN = 38
UT1_CZUJNIK = T2
UT1_PIN = 40
UT1_HISTEREZA = 2
UT2_CZUJNIK = T3
UT2_PIN = 42
UT2_HISTEREZA = 2
UT3_CZUJNIK = T4
UT3_PIN = 44
UT3_HISTEREZA = 2

#TERMOSTAT1_ZADANA = 24
#TERMOSTAT1_HISTEREZA = 1
#TERMOSTAT1_CZUJNIK = TWEW
#TERMOSTAT1_WYJSCIE = CO

PODAJNIK_CZAS_WYRZUTU = 1
PODAJNIK_T_MAX = 60
PODAJNIK_CZAS_PODNOSZENIA = 15
PODAJNIK_T_WYSOKA_POWTORKI = 10
PODAJNIK_T_WYSOKA = 50
PODAJNIK_CZAS_DO_OPROZNIENIA = 3750 #2550
#PODAJNIK_CZAS_PODAJNIKA = 159
PODAJNIK_ALARM_MIN = 10
PODAJNIK_PREDKOSC_PODAWANIA = 1450 #2900

CO_TRYB = ZIMA
CO_ALGORYTM = CIAGLY
CO_CZAS_WLACZ=30
CO_CZAS_WYLACZ=30
#PROG_CO=0
#PROG_CO_ND = 0:00-23:00

CWU_TRYB = WLACZ
CWU_ALGORYTM = CIAGLY
CWU_PRIORYTET = WYLACZ
CWU_ZADANA = 42
CWU_HISTEREZA = 2
CWU_T_MAX = 0

CWU2_TRYB = WYLACZ
CWU2_ALGORYTM = CIAGLY
CWU2_ZADANA =24
CWU2_HISTEREZA = 2
CWU2_CZUJNIK = T1
CWU2_CZUJNIKR = TCWU

CYRKULACJA_ALGORYTM = CYKLICZNY
CYRKULACJA_TMIN = 40
CYRKULACJA_CZAS_WLACZ = 2
CYRKULACJA_CZAS_WYLACZ = 60
#PROG_CYRK_*=5:30-23:00
PROG_CYRK_ND = 8:00-24:00
PROG_CYRK_PO = 6:00-23:00
PROG_CYRK_WT = 6:00-23:00
PROG_CYRK_SR = 6:00-23:00
PROG_CYRK_CZ = 6:00-23:00
PROG_CYRK_PT = 6:00-23:00
PROG_CYRK_SO = 8:00-24:00

NET_IP = 192.168.1.210
NET_MAC = 00E0988070EE
NET_BRAMA = 192.168.1.1
NET_MASKA = 255.255.255.0
NET_DNS = 8.8.8.8
LCD_WYGAS = 6

TZEW = DALLAS[10B6D90F010800A4]
TPOWROT = DALLAS[28A93477910202CF]
TPIEC = DALLAS[28591FDB0A00004A]
TCWU = DALLAS[28208F7997050378]
TPODAJNIK = DALLAS[2873757997050314]
TPODLOGA = DALLAS[28E69DDD0A000028]
TWEW = DALLAS[28452046920902F8]
#T1 = DALLAS[28264D799705036B]      #SOLAR
T2 = DALLAS[28264D799705036B]       #SALON
OFFSET_T2 = -20
T3 = DALLAS[284401779101F2DB]       #LAZIENKA
T4 = DALLAS[2885E277910E028B]       #WIATROLAP
#T5 = DALLAS[282605779101E2F4]
#T6 = DALLAS[2803C64692100227]
TSPALINY = MAX31855[0]

DM_OFFSET = 0
DM_INWERSJA_PWM = 0
DM_MODULACJA = 1
DM_TYP = 1
DM_ROZRUCH_CZAS = 1
DM_ROZRUCH_MOC = 100
DETEKTOR_TESTUJE_BEZPIECZNIK = 1
RTC_ZIMA = 1

koniec `pliku

uzi18, 2021-01-15:
@WojtekGD to chyba z tym sie cos miesza: TWEW_CO_OBNIZONA = 70 sprobuj zakomentowac

WojtekGD, 2021-01-15:
Na razie muszę podgrzać domek tym bardziej, że się ochłodziło. Bartku piec i CO działał prawidłowo tylko termostaty nie sterowały pompą podłogową i elektrozaworami co wyłączyło całkowicie ogrzewanie parteru.

uzi18, 2021-01-15:
Bedziesz mial mozliwosc to zerknij bo tylko tam mieszalem ostatnio

uzi18, 2021-01-21:
@WojtekGD patrzylem w kod i to raczej nie moglo miec wplywu wiec co?

uzi18, 2022-02-25:
@WojtekGD fakt byl blad w odczycie termostatow - poprawione w 0.1.0.130.35

closed 2022-02-25 by uzi18