uzi18/sterownik #114: Nettemp- Lucjan przestaje posyłać dane po restarcie maliny

swiezaks, 2021-05-30:
Cześć. Mam drugą malinę i na niej NT. wszystko poinstalowane bez problemów, lucjan posyła dane do NT, po zrestartowaniu maliny dane z czujników nie docierają do NT, jak ręcznie uruchomię test.py to wszystko znowu działa do czasu kolejnego restartu maliny. Uprawnienia dla plików mam 0777. na końcu pliku rclocal mam exit0 tak jak w instrukcji pisze. 
[image]
[image]

uzi18, 2021-08-09:
@swiezaks 
zaloguj sie przez putty lub uruchom terminal i wpisz: 
screen -r

wklej mi komunikaty

swiezaks, 2021-08-10:
root@pi4:/home/pi# cd /home/pi/lucjan/
root@pi4:/home/pi/lucjan# screen -r
There is no screen to be resumed.

[image]

uzi18, 2021-08-10:
Cos sie zmienilo w nettemp?

swiezaks, 2021-08-11:
Nic się nie zmieniło, aktualizacji juz dawno nie było;) jak ręcznie uruchomię skrypt to wszystko działa do czasu aż zrestartuję malinę, czyli na mój gust jest problem z autostartem "screen".

uzi18, 2021-08-11:
Z logu wynika ze czesc danych sie zapisuje do bazy a czesc nie bo nie widzi baz a za to odpowiada nettemp.
Nie ma w nim nowych urzadzen wykrytych?

swiezaks, 2021-08-11:
docelowo stara malina zostanie wyłączona i na jednej nowej wszystko będzie działało. teraz na czas testów mam dodane kilka czujników ale nie wszystkie. W wolnej chwili jak mi się jeszcze dzisiaj zdalnie z pracy uda to dodam wszystkie czujniki, może to jest problemem?

swiezaks, 2021-08-30:
dodałem wszystkie czujniki ale to nadal nie to, po restarcie maliny o ile dobrze rozumiem nie uruchamia się z automatu test.py w katalogu /run/screen/S-pi/ nie mam żadnego pliku tak jak jest to na drugiej malinie.
Na gitterze w PW podesłałem informacje,

uzi18, 2021-08-31:
```
root@pi4:/home/pi# chown -Rv pi:pi lucjan
właściciel 'lucjan/konfiguracja.py' zachowany jako pi:pi
właściciel 'lucjan/test.py.log' zmieniony z root:root na pi:pi
właściciel 'lucjan/test.py' zachowany jako pi:pi
właściciel 'lucjan/konfiguracja.pyc' zmieniony z root:root na pi:pi
właściciel 'lucjan' zachowany jako pi:pi

```

uzi18, 2021-08-31:
skopiowales z prawami root dlatego nie dzialalo

swiezaks, 2021-09-01:
dziękuję bardzo za pomoc:) wszystko śmiga

closed 2021-09-01 by swiezaks