uzi18/sterownik #116: Nie mogę uruchomić wysyłania danych z Lucjana do Domoticza

swiezaks, 2024-12-25:
Lucjan wersja 0.1.0.130.33s
Domoticz wersja najnowsza 
Version: 2024.7
Build Hash: b317cfab1
Compile Date: 2024-07-13 16:59:31
dzVents Version: 3.1.8
Python Version: 3.11.2 (main, Sep 14 2024, 03:00:30) [GCC 12.2.0]
Active User: admin

konfiguracja.py
```# polaczenie z ethernet shield:
ip_lucjan = '10.10.10.202'
# lub polaczenie przez port szeregory:
#ip_lucjan = '/dev/ttyACM0'
ip_esp = '192.168.2.6'
# 1 - polaczenie przez esp_link:
esp_link = 0
# uwaga Domoticz nie akceptuje polaczenia localhost, nalezy pisac 127.0.0.1
ip_domoticz = '10.10.10.122'
port_domoticz = 8080
idx_start = 1
ip_nettemp = 'localhost'
key_nettemp = 'xxxx'
interwal = 30
interwal_nettemp = 5
interwal_domoticz = 30

idx_zasobnik_cm = 0
idx_zasobnik_procent = 0
```

W domoticzu dodane czujniki wg. instrukcji z wiki ale świecą na czerwono.

Podczas uruchomienia test.py otrzymałem komunikat `multiple exception types must be parenthesized`
w linijce 231 zmieniłem na ` except Exception as e:` i skrypt wystartował.
Logi po uruchomieniu:
```
ETH: czekam na dane
{'thermos': [{'t': 56.1}, {'t': 46.2}, {'t': 35.0}, {'t': 0.8}, {'t': 22.3}, {'t': 18.6}, {'t': 0.0}, {'t': 35.5}, {'t': 0.0}, {'t': 0.0}, {'t': 31.8}, {'t': 20.5}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}], 'wen0': 0, 'wen1': 0, 'podcm': -1, 'podcmp': -1, 'podcz': 28836, 'podczzas': 3924121386, 'podczdo': 41538, 'podgmin': 260, 'pod': 0, 'co': 0, 'cwu1': 1, 'cwu2': 0, 'cyrk': 0, 'ter': 0, 'al': 0, 'time': 661678, 'ver': '0.1.0.130.33s'}

Lucjan RCV:
[56.1, 46.2, 35.0, 0.8, 22.3, 18.6, 0.0, 35.5, 0.0, 0.0, 31.8, 20.5, 0.0, 0.0, 0.0, 0.0]
16
Send NT:
http://localhost/receiver.php?key=xxxx&device=ip&ip=localhost&name=Lucjan_&id=1;2;3;4;5;6;7;8;9;10;11;12;13;14;15;16&type=temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp&value=56.1;46.2;35.0;0.8;22.3;18.6;0.0;35.5;0.0;0.0;31.8;20.5;0.0;0.0;0.0;0.0
ETH: czekam na dane
{'thermos': [{'t': 56.1}, {'t': 46.2}, {'t': 35.0}, {'t': 0.8}, {'t': 22.3}, {'t': 18.6}, {'t': 0.0}, {'t': 35.5}, {'t': 0.0}, {'t': 0.0}, {'t': 31.8}, {'t': 20.5}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}], 'wen0': 0, 'wen1': 0, 'podcm': -1, 'podcmp': -1, 'podcz': 28836, 'podczzas': 3924121386, 'podczdo': 41538, 'podgmin': 260, 'pod': 0, 'co': 0, 'cwu1': 1, 'cwu2': 0, 'cyrk': 0, 'ter': 0, 'al': 0, 'time': 661678, 'ver': '0.1.0.130.33s'}

Lucjan RCV:
[56.1, 46.2, 35.0, 0.8, 22.3, 18.6, 0.0, 35.5, 0.0, 0.0, 31.8, 20.5, 0.0, 0.0, 0.0, 0.0]
16
Send NT:
http://localhost/receiver.php?key=xxxx&device=ip&ip=localhost&name=Lucjan_&id=1;2;3;4;5;6;7;8;9;10;11;12;13;14;15;16&type=temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp&value=56.1;46.2;35.0;0.8;22.3;18.6;0.0;35.5;0.0;0.0;31.8;20.5;0.0;0.0;0.0;0.0
ETH: czekam na dane
{'thermos': [{'t': 56.1}, {'t': 46.2}, {'t': 35.0}, {'t': 0.8}, {'t': 22.3}, {'t': 18.6}, {'t': 0.0}, {'t': 36.0}, {'t': 0.0}, {'t': 0.0}, {'t': 31.8}, {'t': 20.5}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}], 'wen0': 0, 'wen1': 0, 'podcm': -1, 'podcmp': -1, 'podcz': 28836, 'podczzas': 3924121386, 'podczdo': 41538, 'podgmin': 260, 'pod': 0, 'co': 0, 'cwu1': 1, 'cwu2': 0, 'cyrk': 0, 'ter': 0, 'al': 0, 'time': 661678, 'ver': '0.1.0.130.33s'}

Lucjan RCV:
[56.1, 46.2, 35.0, 0.8, 22.3, 18.6, 0.0, 36.0, 0.0, 0.0, 31.8, 20.5, 0.0, 0.0, 0.0, 0.0]
16
Send NT:
http://localhost/receiver.php?key=xxxx&device=ip&ip=localhost&name=Lucjan_&id=1;2;3;4;5;6;7;8;9;10;11;12;13;14;15;16&type=temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp&value=56.1;46.2;35.0;0.8;22.3;18.6;0.0;36.0;0.0;0.0;31.8;20.5;0.0;0.0;0.0;0.0
ETH: czekam na dane
{'thermos': [{'t': 56.1}, {'t': 46.2}, {'t': 35.0}, {'t': 0.8}, {'t': 22.3}, {'t': 18.6}, {'t': 0.0}, {'t': 36.0}, {'t': 0.0}, {'t': 0.0}, {'t': 31.8}, {'t': 20.5}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}], 'wen0': 0, 'wen1': 0, 'podcm': -1, 'podcmp': -1, 'podcz': 28836, 'podczzas': 3924121386, 'podczdo': 41538, 'podgmin': 260, 'pod': 0, 'co': 0, 'cwu1': 1, 'cwu2': 0, 'cyrk': 0, 'ter': 0, 'al': 0, 'time': 661678, 'ver': '0.1.0.130.33s'}

Lucjan RCV:
[56.1, 46.2, 35.0, 0.8, 22.3, 18.6, 0.0, 36.0, 0.0, 0.0, 31.8, 20.5, 0.0, 0.0, 0.0, 0.0]
16
Send NT:
http://localhost/receiver.php?key=xxxx&device=ip&ip=localhost&name=Lucjan_&id=1;2;3;4;5;6;7;8;9;10;11;12;13;14;15;16&type=temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp&value=56.1;46.2;35.0;0.8;22.3;18.6;0.0;36.0;0.0;0.0;31.8;20.5;0.0;0.0;0.0;0.0
ETH: czekam na dane
{'thermos': [{'t': 56.1}, {'t': 46.2}, {'t': 35.0}, {'t': 0.8}, {'t': 22.3}, {'t': 18.6}, {'t': 0.0}, {'t': 36.0}, {'t': 0.0}, {'t': 0.0}, {'t': 31.8}, {'t': 20.5}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}], 'wen0': 0, 'wen1': 0, 'podcm': -1, 'podcmp': -1, 'podcz': 28836, 'podczzas': 3924121386, 'podczdo': 41538, 'podgmin': 260, 'pod': 0, 'co': 0, 'cwu1': 1, 'cwu2': 0, 'cyrk': 0, 'ter': 0, 'al': 0, 'time': 661679, 'ver': '0.1.0.130.33s'}

Lucjan RCV:
[56.1, 46.2, 35.0, 0.8, 22.3, 18.6, 0.0, 36.0, 0.0, 0.0, 31.8, 20.5, 0.0, 0.0, 0.0, 0.0]
16
Send NT:
http://localhost/receiver.php?key=xxxx&device=ip&ip=localhost&name=Lucjan_&id=1;2;3;4;5;6;7;8;9;10;11;12;13;14;15;16&type=temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp&value=56.1;46.2;35.0;0.8;22.3;18.6;0.0;36.0;0.0;0.0;31.8;20.5;0.0;0.0;0.0;0.0
ETH: czekam na dane
{'thermos': [{'t': 56.1}, {'t': 46.2}, {'t': 35.0}, {'t': 0.8}, {'t': 22.3}, {'t': 18.6}, {'t': 0.0}, {'t': 35.7}, {'t': 0.0}, {'t': 0.0}, {'t': 31.8}, {'t': 20.5}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}], 'wen0': 0, 'wen1': 0, 'podcm': -1, 'podcmp': -1, 'podcz': 28836, 'podczzas': 3924121386, 'podczdo': 41538, 'podgmin': 260, 'pod': 0, 'co': 0, 'cwu1': 1, 'cwu2': 0, 'cyrk': 0, 'ter': 0, 'al': 0, 'time': 661679, 'ver': '0.1.0.130.33s'}

Lucjan RCV:
[56.1, 46.2, 35.0, 0.8, 22.3, 18.6, 0.0, 35.7, 0.0, 0.0, 31.8, 20.5, 0.0, 0.0, 0.0, 0.0]
16
Send NT:
http://localhost/receiver.php?key=xxxx&device=ip&ip=localhost&name=Lucjan_&id=1;2;3;4;5;6;7;8;9;10;11;12;13;14;15;16&type=temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp&value=56.1;46.2;35.0;0.8;22.3;18.6;0.0;35.7;0.0;0.0;31.8;20.5;0.0;0.0;0.0;0.0
ETH: czekam na dane
{'thermos': [{'t': 56.1}, {'t': 46.2}, {'t': 35.0}, {'t': 0.8}, {'t': 22.3}, {'t': 18.6}, {'t': 0.0}, {'t': 35.7}, {'t': 0.0}, {'t': 0.0}, {'t': 31.8}, {'t': 20.5}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}], 'wen0': 0, 'wen1': 0, 'podcm': -1, 'podcmp': -1, 'podcz': 28836, 'podczzas': 3924121386, 'podczdo': 41538, 'podgmin': 260, 'pod': 0, 'co': 0, 'cwu1': 1, 'cwu2': 0, 'cyrk': 0, 'ter': 0, 'al': 0, 'time': 661679, 'ver': '0.1.0.130.33s'}

Lucjan RCV:
[56.1, 46.2, 35.0, 0.8, 22.3, 18.6, 0.0, 35.7, 0.0, 0.0, 31.8, 20.5, 0.0, 0.0, 0.0, 0.0]
16
Send NT:
http://localhost/receiver.php?key=xxxx&device=ip&ip=localhost&name=Lucjan_&id=1;2;3;4;5;6;7;8;9;10;11;12;13;14;15;16&type=temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp&value=56.1;46.2;35.0;0.8;22.3;18.6;0.0;35.7;0.0;0.0;31.8;20.5;0.0;0.0;0.0;0.0
ETH: czekam na dane
{'thermos': [{'t': 56.1}, {'t': 46.2}, {'t': 35.0}, {'t': 0.8}, {'t': 22.3}, {'t': 18.6}, {'t': 0.0}, {'t': 35.7}, {'t': 0.0}, {'t': 0.0}, {'t': 31.8}, {'t': 20.5}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}], 'wen0': 0, 'wen1': 0, 'podcm': -1, 'podcmp': -1, 'podcz': 28836, 'podczzas': 3924121386, 'podczdo': 41538, 'podgmin': 260, 'pod': 0, 'co': 0, 'cwu1': 1, 'cwu2': 0, 'cyrk': 0, 'ter': 0, 'al': 0, 'time': 661679, 'ver': '0.1.0.130.33s'}

Lucjan RCV:
[56.1, 46.2, 35.0, 0.8, 22.3, 18.6, 0.0, 35.7, 0.0, 0.0, 31.8, 20.5, 0.0, 0.0, 0.0, 0.0]
16
Send NT:
http://localhost/receiver.php?key=xxxx&device=ip&ip=localhost&name=Lucjan_&id=1;2;3;4;5;6;7;8;9;10;11;12;13;14;15;16&type=temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp;temp&value=56.1;46.2;35.0;0.8;22.3;18.6;0.0;35.7;0.0;0.0;31.8;20.5;0.0;0.0;0.0;0.0
ETH: czekam na dane
^CTraceback (most recent call last):
  File "/home/pi/lucjan/test.py", line 100, in <module>
    j    = response.read().decode("utf-8")
           ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/http/client.py", line 478, in read
    s = self.fp.read()
        ^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/socket.py", line 706, in readinto
    return self._sock.recv_into(b)
           ^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt

```

i teraz nie wiem czy problem jest z ustawieniem domoticza czy coś innego.
Proszę o podpowiedź.

uzi18, 2024-12-26:
Masz racje problem jest z pythonem. 
W nowej wersji cos zmienili.

swiezaks, 2024-12-26:
Czyli mam zrobić downgrade pythona? jeżeli tak to na której wersji na pewno będzie działać? Chyba że planujesz jakąś poprawkę na najnowszą wersję pythona?
Pozdrawiam i dzięki za odpowiedź.

uzi18, 2024-12-26:
Tymczasowo uruchom na python2, mam gdzies poprawke do python3

swiezaks, 2024-12-26:
oki, dzięki, zabieram się za kombinowanie :)))

uzi18, 2024-12-26:
Daj cynk czy dziala

uzi18, 2024-12-26:
Jednak wrzucilem poprawke do python3 ktora wersje sciagnales?

uzi18, 2024-12-26:
Najnowszy soft lucjana to z koncowka 35

swiezaks, 2024-12-26:
Co do  softu lucjana to wiem, ale padła mi karta W5100 i nie mogłem się połączyć z www, w ferworze walki wgrałem wersję jakąś starszą którą miałem na laptopie do programowania ale nic to nie dało, po wymianie w5100 wszystko ruszyło to już nie chciałem nic kombinować bo były mrozy wtedy;)
Skoro dodałeś poprawkę (dzięki) to nie będę kombinował bo za pół roku albo rok nie będę pamiętał co robiłem żeby zadziałało:)
aktualnie mam wersję Python 3.11.2.
Zaraz zrobię cały nowy system i na nowo wszystko zainstaluję bo już nieźle kombinowałem żeby to uruchomić i sam nie wiem co jest dobre a co złe:)
Widziałem, że jeszcze coś robiłeś chyba z poziomem zasobnika jak dobrze rozumiem. Powiedz mi proszę jaki dokładnie czujnik poziomu masz zastosowany bo mnie się go nie udało uruchomić (jakaś wersja trefna z aliexpress).

uzi18, 2024-12-26:
Masz to w plikach napisane. 
Nie wiem czy po tylu latach jest jeszcze dostepny w tej samej formie. 
Musialbys poszukac i wkleic tu linka.
Ściągnij sobie najnowszy test.py powinien byc ok, chyba ze github ci podrzuca starsza wersje.
Moze jeszcze cos niedomaga z python3, nie wszystko bylem w stanie przetestowac.

uzi18, 2024-12-26:
Zakomentuj ip_nettemp skoro nie używasz

swiezaks, 2024-12-26:
na ten moment aktualizuję system, wchodząc na https://github.com/uzi18/sterownik/tree/master/arduino/Domoticz
widać aktualizacja 3 tygodnie temu i na tym mi nie działało. Nie wiem czy zmianę którą robiłeś dzisiaj powinno być widać od razu?
[image]

uzi18, 2024-12-26:
Ok to sprawdz z ta konfiguracja tak jak napialem dodaj # przed tymi ustawieniami w konfiguracji

swiezaks, 2024-12-26:
konfiguracja.py wygląda tak
```
  GNU nano 7.2                     konfiguracja.py                              
ip_lucjan = '10.10.10.202'
# lub polaczenie przez port szeregory:
#ip_lucjan = '/dev/ttyACM0'
#ip_esp = '192.168.2.6'
# 1 - polaczenie przez esp_link:
#esp_link = 0
# uwaga Domoticz nie akceptuje polaczenia localhost, nalezy pisac 127.0.0.1
ip_domoticz = '10.10.10.204'
port_domoticz = 8080
idx_start = 1
#ip_nettemp = 'localhost'
#key_nettemp = 'xxxx'
interwal = 30
#interwal_nettemp = 5
interwal_domoticz = 30

idx_zasobnik_cm = 0
idx_zasobnik_procent = 0

```

po uruchomieniu test.py w terminalu mam tak
```
Send DM:
http://10.10.10.204:8080/json.htm?type=command&param=udevice&idx=1&nvalue=0&svalue=54.9
ETH: czekam na dane
{'thermos': [{'t': 54.9}, {'t': 47.4}, {'t': 26.1}, {'t': -4.9}, {'t': 22.7}, {'t': 52.2}, {'t': 0.0}, {'t': 39.0}, {'t': 0.0}, {'t': 0.0}, {'t': 40.1}, {'t': 21.0}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}, {'t': 0.0}], 'wen0': 42, 'wen1': 42, 'podcm': -1, 'podcmp': -1, 'podcz': 36427, 'podczzas': 3924113795, 'podczdo': 41538, 'podgmin': 260, 'pod': 1, 'co': 1, 'cwu1': 0, 'cwu2': 0, 'cyrk': 0, 'ter': 0, 'al': 0, 'time': 780717, 'ver': '0.1.0.130.33s'}

Lucjan RCV:
[54.9, 47.4, 26.1, -4.9, 22.7, 52.2, 0.0, 39.0, 0.0, 0.0, 40.1, 21.0, 0.0, 0.0, 0.0, 0.0]
16
Send DM:
http://10.10.10.204:8080/json.htm?type=command&param=udevice&idx=1&nvalue=0&svalue=54.9
ETH: czekam na dane
```
natomiast w domoticzu nic się nie pojawia pomimo odczekania nawet kilku minut.

swiezaks, 2024-12-27:
a w logu sypie
```
2024-12-27 00:36:18,933 ERROR <module>(68) Start
2024-12-27 00:36:19,149 ERROR <module>(232) Exception
Traceback (most recent call last):
  File "/home/pi/lucjan/test.py", line 200, in <module>
    response = urlopen(adr)
               ^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 525, in open
    response = meth(req, response)
               ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 634, in http_response
    response = self.parent.error(
               ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 563, in error
    return self._call_chain(*args)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 496, in _call_chain
    result = func(*args)
             ^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 643, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 401: Unauthorized
2024-12-27 00:36:19,379 ERROR <module>(232) Exception
Traceback (most recent call last):
  File "/home/pi/lucjan/test.py", line 200, in <module>
    response = urlopen(adr)
               ^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 525, in open
    response = meth(req, response)
               ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 634, in http_response
    response = self.parent.error(
               ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 563, in error
    return self._call_chain(*args)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 496, in _call_chain
    result = func(*args)
             ^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 643, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 401: Unauthorized
2024-12-27 00:36:19,621 ERROR <module>(232) Exception
Traceback (most recent call last):
  File "/home/pi/lucjan/test.py", line 200, in <module>
    response = urlopen(adr)
               ^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 525, in open
    response = meth(req, response)
               ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 634, in http_response
    response = self.parent.error(
               ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 563, in error
    return self._call_chain(*args)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 496, in _call_chain
    result = func(*args)
             ^^^^^^^^^^^
  File "/usr/lib/python3.11/urllib/request.py", line 643, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 401: Unauthorized
2024-12-27 00:36:19,942 ERROR <module>(232) Exception
Traceback (most recent call last):
```

uzi18, 2024-12-27:
Masz jak wół blad o braku autoryzacji w domoticz. 
Dodaj IP urzadzenia na ktorym ten skrypt dziala w domoticz jako dozwolone urządzenie bez autoryzacji.

uzi18, 2024-12-27:
Trzeba bedzie dodac do wiki co zrobic w domoticz jak jest blad 401

swiezaks, 2024-12-27:
Udało się:) faktycznie po dodaniu adresu IP w konfiguracja/ustawienia/security w polu Trusted Networks (no username/password): wprowadzenie adresu IP lucjana sprawiło że skrypt wystartował i dane pojawiły się w Domoticzu- SUKCES.
Oczywiście jak to już mam w zwyczaju nie wszystko działa chociaż innym działa.
Z racji tego iż mam wgrany obraz z desktopem (pi4 4GBram) to nie ma u mnie pliku /etc/rc.local i teraz pytanie- 
dodałem do crontab 
```
@reboot python3 /home/pi/lucjan/test.py &
'''
i wszystko działa, mogę tak zostawić czy są bardziej eleganckie sposoby?
Domoticza używam pierwszy raz i jeszcze długa droga przede mną  ale pierwsze koty za płoty:)

uzi18, 2024-12-31:
Moze byc, ogolnie to domoticz jest do bani, HA jest trudniejsze w początkowej konfiguracji ale za to dziala lepiej i stabilniej

swiezaks, 2025-01-05:
Dzięki za pomoc, domoticz ma służyć za most pomiędzy lucjanem a MagicMirror2. plus jest taki, że domoticz mało zasobów potrzebuje w porównaniu do HA i mogę go używać na pi4 z desktopem i odtwarzaniem filmów na jednym urządzeniu.
@uzi18 jak możesz to zerknij na jeszcze dwa issue który otworzyłem przed chwila:) 
Dzięki za wsparcie:)

swiezaks, 2025-01-05:
temat wyjaśniony i działa wysyłanie danych z lucjana do domoticza.

closed 2025-01-05 by swiezaks